ihedvall/mdflib #1: Using Qt instead of boost

Murmele, 2022-09-21:
Hello,

the [labplot team](https://labplot.kde.org/) is thinking about supporting mdf files in their application. Labplot is using Qt and we would like to get an additional dependency like boost and so the idea would be to replace boost by Qt equivalents. In this case I would create a library interface (source, header file) for boost and qt and in the cmake file the correct files will be selected. So that means your mdf lib depend only on that library interface files and anymore directly on boost.

Mostly locale and endian have to be replaced.
The current comparsion and alternative Qt function selection is done here: https://invent.kde.org/education/labplot/-/issues/293

What do you think about this? I will do the implementation of this procedure.

ihedvall, 2022-09-21:
Hello

The MDFLIB repository includes a library and a sub-project MdfViewer (wxWidgets). The viewer application can easily be moved to its own repository.

The library is dependent on 3:rd party libraries as boost, expat, utillib and zlib.

The boost library solve the big/little endian byte order issues. Instead of involving Qt, I suggest that the byte order stuffing is replaced by plain C code instead. This will solve the main boost dependency.

The UtilLib constains a lot of XML (expat) and time conversion routines. It also include error logging functions. The time and logging functions can be included (copied) into the library.

Remains from the UtilLib is the XML (expat) dependency. XML is heavily used in the MDF4 Meta data block (MD block).

The ZLIB is used for compressing/decompressing the MDF4 data blocks. I don't see that this dependecy can be removed.

### Conclusion
I can do the above changes removing dependecies of the utillib and boost library. Remain dependency of expat and zlib.
Is this an acceptable solution?

ihedvall, 2022-09-21:
FYI
I have started to remove dependecies. Boost and utilib is OK. Some problems with XML and Expat. I'm 1-2 days left.

Murmele, 2022-09-21:
Hello. Thank you very much for the changes! I think boost is the biggest dependency which might be great to get rid of it. Zlib is used also for hdf5 so it is already a dependency of labplot so it should not be any problem. I think you don't have to move the viewer out, because when building without the viewer then wxwidgets and so on is not needed, which makes it fine. 
If I can help you with something I would be happy to do. I will let you know tomorrow about the changes for building on linux

Murmele, 2022-09-21:
Are you using any codestyle template like clang-format?

ihedvall, 2022-09-22:
I have removed the utillib dependency in mdflib. UtilLib uses boost. MdfLib should only depend on EXPAT and ZLIB.
I'm using the JetBrain CLION (Editor) and TeamCity (CI) with local agents. The latter due to the unit tests require local MDF files that cannot be exposed for external users.

I'm using the Google C++ coding standard. The Editor has built-in clang which is good enough so far

Murmele, 2022-09-22:
> I have removed the utillib dependency in mdflib. UtilLib uses boost. MdfLib should only depend on EXPAT and ZLIB. I'm using the JetBrain CLION (Editor) and TeamCity (CI) with local agents. The [...]

Did you push the changes?

ihedvall, 2022-09-22:
No. I'm removing the boost dependency as well.

Murmele, 2022-09-22:
Ah ok thanks. Thank you for your work and helping to get mdflib cross compiled

ihedvall, 2022-09-22:
I have checked in the changes that should remove the boost and utillib dependencies. We need to merge in the changes from you. I have added you as collaborator.

Best Regards
Ingemar Hedvall

Murmele, 2022-09-24:
Thank you for your effort! I can close this one now

closed 2022-09-24 by Murmele