ihedvall/mdflib #2: error: could not convert ‘0’ from ‘int’ to ‘fpos_t’

Murmele, 2022-09-21:
```
mdflib/src/iblock.h:131: Fehler: could not convert ‘0’ from ‘int’ to ‘fpos_t’
In file included from mdflib/src/pr3block.h:8,
                 from mdflib/src/pr3block.cpp:8:
mdflib/src/iblock.h:131:27: error: could not convert ‘0’ from ‘int’ to ‘fpos_t’
  131 |   fpos_t file_position_ = 0;       ///< 64-bit file position.
      |                           ^
      |                           |
      |                           int
```

If I am looking into the definition of fpos_t it is a struct and not just a number
[image]

Murmele, 2022-09-21:
The problem is that fpos_t is defined differently on different platforms

Murmele, 2022-09-21:
According to the reference, fpos_t is used incorrectly: https://cplusplus.com/reference/cstdio/fpos_t/

only using fgetpos and fsetpos are defined functions

ihedvall, 2022-09-21:
This is a bug. The MDF3 uses 32-bit (uint32_t) file position while the MDF4 uses 64-bit file position (int64_t). I had problem earlier with setting and getting file position which depends on the compiler.

The solution is to remove all fpos_t and fgetpos... from the source code and isolate all that code in one place. 

The iblock.h/cpp shall do all file position setting and getting. I can fix the bug but I suspect this is not the last bug related to this.

ihedvall, 2022-09-21:
I have isolated the fpos_t problem to the Iblock.cpp file functions Get/SetFilePosition(). The current solution may work but I don't have all platforms available so some ifdef might be required in these functions. Is it possible for you to test?

Murmele, 2022-09-21:
There are quite a lot ifdes to add but it is going forward

ihedvall, 2022-09-21:
Note that I checked in changes. All fpos_t related thing should be in 2 functions in src/iblock.cpp.

Murmele, 2022-09-21:
Thanks I will rebase tomorrow my branch and go on with replacing windows specific functions by c++ std or ifdefs.

closed 2022-09-24 by Murmele