ihedvall/mdflib #5: Reader example not working

Murmele, 2022-09-24:
Currently I am trying to get the example working, but for me it is not working:

[image]

ihedvall, 2022-09-24:
Yes it's a bug. It should be obs-> not sub->

ihedvall, 2022-09-24:
Many unit test require some test file(s). Unfortunately, most files are proprietary of a company. I am also missing MDF3 files.  

This is a common problem. so normally I solve this by checking if the remote direcctory exist and if not, don't run the test (well they always succeed).

I don't know if this is the best solution.

Murmele, 2022-09-24:
> 

I think it should at least compile. I made a proposal in #7 how to achieve this

ihedvall, 2022-09-24:
Not perfect solution but it doesn't break anything

closed 2022-09-24 by ihedvall