ihedvall/mdflib #8: Showing linked information in tree

Murmele, 2022-09-25:
Is it possible to directly showing the metadata directly in the table instead of just (link to unit?)
[image]

ihedvall, 2022-09-25:
Well maybe. I only have one line to show data in or at maybe multiline may work. I could check if the TXT or the MD block only consist of a "one liner" text, then showing the contents instead of the link. I can see what can be done

Murmele, 2022-09-25:
Or maybe when clicking on the link showing the content in a dialog?

ihedvall, 2022-09-25:
A double click should either show pop-up or set the tree-list to the block. Note that the link may me NULL i.e. pointing on nothing.

ihedvall, 2022-09-25:
I have checked in the changes so the TX/MD blocks "main" text are show in the description field. 

Still in the GUI, is the bitmap problem. Windows uses BITMAP format while Unix and MacOS uses some other format (PNG?). I'm setting up an Ubuntu computer for development but it take some days before it works.

closed 2022-09-25 by ihedvall