ihedvall/mdflib #11: Get windows and macos CI build running

Murmele, 2022-10-04:


ihedvall, 2022-10-08:
The main problem seems to be the source_location that MSVC and GCC uses while the CLANG doesn't support. There is several alternative solutions on this issue,

1. Use MSVC and GCC as compiler.
2. Rewrite the MdfLogStream so it doesn't use the source_location. This requires a fork or #ifdef:s inside the streamer class. 
3. Wait until CLANG version support C++ 20 libraries.

Murmele, 2022-10-08:
I think changing to GCC is fine. Neverthless to compile then with labplot I have to check if it is possible to build for all systems

ihedvall, 2022-10-08:
I have changed the cmake to use FetchContent build instead of FindPackage for expat in the utillib. This seems to working as the library is small. The same should work for the mdflib build. Assume that the zlib also should be build  the FetchContent way.

Murmele, 2022-10-08:
is source_location the only c++20 feature which is used?

ihedvall, 2022-10-08:
No. 
Also ranges are used and most likely some other features. Labplot require C++11 and C99.

Murmele, 2022-10-08:
Yes currently c++11 and c99 are used. I got labplot compiled with c++20 I have to talk with them If we should upgrade the project to c++20.

ihedvall, 2022-10-08:
I might be a good idea to upgrade to modern standard. Note that most C++20 code lines can be changed to C++ 11 standard. It is some days work but should work. I let you decide. It depends how old Labplot compilers should support. 

Do you know any tool that can check for c++ 11/20 issues?

Murmele, 2022-10-08:
You mean some quality checks?
kde is using something in their CI pipeline: https://invent.kde.org/education/labplot/-/merge_requests/157
[image]

But currently we don't consider it.
I think it is cppcheck: http://cppcheck.net/

ihedvall, 2023-02-17:
I have some idea to start using the vcpkg for loading the third-party libraries. Any recommendations regarding this approach?

closed 2023-08-17 by Simplxss