ihedvall/mdflib #14: About write MDFfile

nealwang123, 2023-04-12:
There is a requirement to record data in MDF files in the standardization record function for automotive components.Would it have been possible to support it earlier

ihedvall, 2023-04-12:
There already exist some write support in the library so finalize it is a relative small work. However there are a large number of way the sample values may be stored. MDF3 support writing of values but in MDF4 there are a lot of features added. Adding support of all the features is several months work. I need some help regarding this by the end-user of the library.

Examples:

- CAN loggers store raw CAN data bytes in the MDF file. To parse out the values, you need to first parse the DBC file and the parse the raw data bytes into values. There exist some flavours here.

- Another way is to store the enginnering values (no conversion CC block). Values with variable length as strings, need special handling.

- The intended way of storing fixed size channel values require conversion blocks which require some skills to figure out. Pressure sensors uses individual (calibrated) conversion while TK sensors uses map conversions. 

Well, I need help to define the input and the output of the MDF writer, simply some requirements.

An initial proposal from me is to implement one CAN logger writer interface and one simple writer interface that store compressed samples (DZ) blocks. Any ideas?

nealwang123, 2023-04-13:
I'm just starting to get into the mdf file format, and all I can think of is if we can define each module as a class first for easy extension later.

closed 2023-05-10 by ihedvall