ihedvall/mdflib #15: Some Question in using lib

young147, 2023-05-15:
Hi! I am tring to use your lib to write mf4 file, but a problem I met was that there is no data inside mf4 file when I use CANape open this file, and the code generated mf4 file is refer to the test/testwrite.cpp, I would be grateful if you can help me.

ihedvall, 2023-05-15:
The unit tests generates many mdf files and some of them are actually empty (no data). The unit test Mdf4Float (float.mf4) should 
have 100 samples inside.  Note that write support is just some days old. I don't have CANape but I can try to download the demo version.

young147, 2023-05-15:
> The unit tests generates many mdf files and some of them are actually empty (no data). The unit test Mdf4Float (float.mf4) should have 100 samples inside. Note that write support is just some days old. I don't have CANape but I can try to download the demo version.
Thanks, I  just noted that you updated the unitest code, I will try to fix the problem according to your code.

ihedvall, 2023-05-15:
I downloaded the CANape demo version and it plots the float.mf4 even if the x-axis is wrong as the file doesn't have a time channel. You can also the MDFViewer executable if you just want to check the values. 

I'm working with the Doxygen documentation and an installation kit (Windows). I'm ready with that later this week. I will set a release at this point.

young147, 2023-05-16:
> I downloaded the CANape demo version and it plots the float.mf4 even if the x-axis is wrong as the file doesn't have a time channel. You can also the MDFViewer executable if you just want to check [...]

Thanks for your help, I have already write MF4 file successful include the situation that multi-channel groups, it's very grateful for your work.

ihedvall, 2023-05-21:
I have published the version 2.0 of the MDF library.

I'm not using the write functionality in my other library/projects so if you find any problems or want any improvments, don't hesitate to give me a note.

Best Regards
Ingemar Hedvall

closed 2023-05-21 by ihedvall