ihedvall/mdflib #23: About CnFlag and ChannelType

YuFanXing, 2023-07-23:
In *cn4block.cpp*
```cpp {.line-numbers}
void Cn4Block::PrepareForWriting(size_t offset) {
  bit_offset_ = 0;
  byte_offset_ = offset;
  data_list_.clear(); // Temporary storage of signal data (SD)
  data_map_.clear();
  // The bit count may be set by the user.
  bool use_index = false;
  switch (Type()) {
    case ChannelType::MaxLength:
    case ChannelType::Sync:
    case ChannelType::Master:
    case ChannelType::FixedLength:
      // Length fixed below or by user
      // Always store invalid bit
      flags_ |= CnFlag::InvalidValid;
      break;

    case ChannelType::VariableLength:
      // Store unsigned 32/64-bit index to variable block
      flags_ |= CnFlag::InvalidValid;
      if (bit_count_ == 0) {
        bit_count_ = 8 * 8;
      }
      return;

    case ChannelType::VirtualData:
    case ChannelType::VirtualMaster:
      // Sample index * CC sets the channel value. No valid bits
      bit_count_ = 0;
      flags_ &= ~CnFlag::InvalidValid;
      return;
  }
  // ...
}
```
It's not necessary that add *InvalidValid* to flags. In fact most *FixedLength* value don't store invalid bit, and *Master* value even can't contain invalid bit.

ihedvall, 2023-07-23:
It's OK. I can remove this flag. I can fix this. It's relative easy (2 places) but some of the unit tests need to be updated as well,

ihedvall, 2023-07-23:
Fixed so the user selects the invalid flag usage.

closed 2023-07-23 by ihedvall