ihedvall/mdflib #26: Cannot read string value written by MDFWriter

Simplxss, 2023-07-26:
```c#
TestWriter()
{
    var Writer = new MdfWriter(MdfWriterType.Mdf4Basic);
    Writer.Init(TestFile4);
    var dg = Writer.CreateDataGroup();
    var cg = dg.CreateChannelGroup();
    var cn = cg.CreateChannel();
    cn.Name = "String";
    cn.Description = "string";
    cn.Type = ChannelType.FixedLength;
    cn.DataType = ChannelDataType.StringAscii;
    cn.DataBytes = 10;
    Writer.InitMeasurement();
    Writer.StartMeasurement((ulong)(DateTimeOffset.Now.ToUnixTimeMilliseconds() * 1000000));

    // i from 0 to 49
    cn.SetChannelValue(i.ToString());

    Writer.StopMeasurement((ulong)(DateTimeOffset.Now.ToUnixTimeMilliseconds() * 1000000));
    Writer.FinalizeMeasurement();
}
TestReader2()
{
    var Reader = new MdfReader(TestFile4);
    Reader.ReadEverythingButData();
    // ... create channel observer
    Reader.ReadData(group);
    string channel_value = ""; // Channel value (no scaling)
    string eng_value = ""; // Engineering value
    (MdfChannelObserver)item.GetChannelValueAsString(i, ref channel_value);
    (MdfChannelObserver)item.GetEngValueAsString(i, ref eng_value);
    Console.WriteLine($"channel_value = {channel_value}, eng_value= {eng_value}");
}
```
Detail code see `TestWriter` & `TestReader2` functions in google test.

I viewed the output file and found the string seems writting correctly(?).
[image]

But all values in `value_list_` of ChannelObserver are "" when reading.
It seems that there are some problems in `Reader.ReadData()`.
I have tried to solve it, but It's a little diffcult for me to fully understand how it works...

ihedvall, 2023-07-26:
In general, you shouldn't use fixed length string. MDF file format and strings, doesn't go well together.

If you change the string channels data type to a variable length type and set data size to 4 or 8 bytes (this is the index size). See also unit test TestWrite::StringData.

However fixed length strings should work. I do a unit test for this and check what's the problem is.

ihedvall, 2023-07-26:
I think the problem is solved. It was working for the first channel but not for the next channels. You can verify in .NET world.

Simplxss, 2023-07-26:
It works well now,  but I'm confused that why it needs to convent ASCII to UTF-8 encoding. In my view that UTF-8 contains ASCII tables.

ihedvall, 2023-07-26:
I think ASCII also include the 0x10-0xFF characters. I'm using UTF8 if possible.

I have a question regarding timestamp value. C++ is still missing a timestamp value but when using Protobuf/gRPC. I'm using a protobuf version. Do C#/.NET have a standard way of expressing an absolute time which is better than using the nano-second definition from C++?

Simplxss, 2023-07-26:
Sorry, in fact I'm not familiar with C#/.NET. But I don't think there is an appropriate way to process time better than timestamp. DateTime in C# could only handle 1/1000 ms at most.

Simplxss, 2023-07-26:
Yes, DateTime will be impoved in .NET 7, but we can't trial it before .NET 8 releasing...