ihedvall/mdflib #32: Unable to build lib for Ubuntu 20.04 x86_64

edmendezcr, 2023-08-02:
Hi there,

I'm not able to build the library for Ubuntu 20.04 x86_64 so I wonder if you have seen the same problem. Seems like the issue is `template argument deduction/substitution failed`:

```
In file included from /home/ubuntu/mdflib/include/mdf/ichannelconversion.h:15,
                 from /home/ubuntu/mdflib/src/ichannelconversion.cpp:6:
/usr/include/c++/9/variant:1351:2: note: candidate: ‘template<long unsigned int _Np, class _Up, class ... _Args, class _Tp, class> constexpr std::variant<_Types>::variant(std::in_place_index_t<_Np>, std::initializer_list<_Up>, _Args&& ...)’
 1351 |  variant(in_place_index_t<_Np>, initializer_list<_Up> __il,
      |  ^~~~~~~
/usr/include/c++/9/variant:1351:2: note:   template argument deduction/substitution failed:
In file included from /usr/include/x86_64-linux-gnu/c++/9/bits/c++allocator.h:33,
                 from /usr/include/c++/9/bits/allocator.h:46,
                 from /usr/include/c++/9/string:41,
                 from /usr/include/c++/9/stdexcept:39,
                 from /usr/include/c++/9/optional:38,
                 from /home/ubuntu/mdflib/include/mdf/ichannelconversion.h:11,
                 from /home/ubuntu/mdflib/src/ichannelconversion.cpp:6:
/usr/include/c++/9/ext/new_allocator.h:146:4: note:   mismatched types ‘std::in_place_index_t<_Idx>’ and ‘long long unsigned int’
  146 |  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

```
ubuntu@ubuntu:~/mdflib/build/release$ cmake -G Ninja -DCMAKE_BUILD_TYPE=Release ../..
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ZLIB: /usr/local/lib/libz.a (found version "1.2.13")  
-- ZLIB Found (Try 1): TRUE
-- ZLIB Version: 1.2.13
-- ZLIB Include Dirs: /usr/local/include
-- ZLIB Libraries: /usr/local/lib/libz.a
-- Found EXPAT: /usr/lib/x86_64-linux-gnu/libexpat.so (found version "2.2.9") 
-- Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES wxWidgets_INCLUDE_DIRS) 
-- Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES wxWidgets_INCLUDE_DIRS) 
wxWidgets requested but not found.
-- wxWidgets Find Style: unix
-- wxWidgets Lib Dir: 
-- wxWidgets Config Exe : wxWidgets_CONFIG_EXECUTABLE-NOTFOUND
-- wxWidgets Cmake Root Path : /wxwidgets/master
-- wxWidgets Root: 
-- wxWidgets Found: FALSE
-- wxWidgets Include Dirs: 
-- wxWidgets Library Dirs: 
-- wxWidgets Libraries: 
-- wxWidgets Include File: /home/ubuntu/.local/lib/python3.8/site-packages/cmake/data/share/cmake-3.27/Modules/UsewxWidgets.cmake
-- wxWidgets Definitions: 
-- wxWidgets Debug Definitions: 
-- wxWidgets CXX Flags: 
-- wxWidgets Found: FALSE
-- wxWidgets Include Dirs: 
-- wxWidgets Library Dirs: 
-- wxWidgets Libraries: 
-- wxWidgets Include File: /home/ubuntu/.local/lib/python3.8/site-packages/cmake/data/share/cmake-3.27/Modules/UsewxWidgets.cmake
-- wxWidgets Definitions: 
-- wxWidgets Debug Definitions: 
-- wxWidgets CXX Flags: 
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- GTest Found: FALSE
-- GTest Include Dirs: 
-- GTest Libraries: 
-- GTest Main Libraries: 
-- GTest Both Libraries: 
-- Could NOT find Doxygen (missing: mscgen dia) (found version "1.8.17")
-- Doxygen Found: NO
-- Doxygen Version: 1.8.17
-- MDF Target Includes: include/mdf/cryptoutil.h;include/mdf/etag.h;include/mdf/iattachment.h;include/mdf/iblock.h;include/mdf/ichannel.h;include/mdf/ichannelarray.h;include/mdf/ichannelconversion.h;include/mdf/ichannelgroup.h;include/mdf/ichannelhierarchy.h;include/mdf/ichannelobserver.h;include/mdf/idatagroup.h;include/mdf/ievent.h;include/mdf/ifilehistory.h;include/mdf/iheader.h;include/mdf/imetadata.h;include/mdf/isampleobserver.h;include/mdf/isourceinformation.h;include/mdf/mdffactory.h;include/mdf/mdffile.h;include/mdf/mdfhelper.h;include/mdf/mdflogstream.h;include/mdf/mdfreader.h;include/mdf/mdfwriter.h;include/mdf/samplerecord.h;include/mdf/zlibutil.h
-- Configuring done (0.6s)
-- Generating done (0.0s)
-- Build files have been written to: /home/ubuntu/mdflib/build/release

ubuntu@ubuntu:~/mdflib/build/release$ ninja
[13/81] Building CXX object CMakeFiles/mdf.dir/src/ichannelconversion.cpp.o
FAILED: CMakeFiles/mdf.dir/src/ichannelconversion.cpp.o 
/usr/bin/c++ -DXML_STATIC -I/home/ubuntu/mdflib/include -I/expat/lib -O3 -DNDEBUG -std=gnu++17 -MD -MT CMakeFiles/mdf.dir/src/ichannelconversion.cpp.o -MF CMakeFiles/mdf.dir/src/ichannelconversion.cpp.o.d -o CMakeFiles/mdf.dir/src/ichannelconversion.cpp.o -c /home/ubuntu/mdflib/src/ichannelconversion.cpp
In file included from /usr/include/x86_64-linux-gnu/c++/9/bits/c++allocator.h:33,
                 from /usr/include/c++/9/bits/allocator.h:46,
                 from /usr/include/c++/9/string:41,
                 from /usr/include/c++/9/stdexcept:39,
                 from /usr/include/c++/9/optional:38,
                 from /home/ubuntu/mdflib/include/mdf/ichannelconversion.h:11,
                 from /home/ubuntu/mdflib/src/ichannelconversion.cpp:6:
/usr/include/c++/9/ext/new_allocator.h: In instantiation of ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = std::variant<long unsigned int, double>; _Args = {long long unsigned int}; _Tp = std::variant<long unsigned int, double>]’:
/usr/include/c++/9/bits/alloc_traits.h:483:4:   required from ‘static void std::allocator_traits<std::allocator<_CharT> >::construct(std::allocator_traits<std::allocator<_CharT> >::allocator_type&, _Up*, _Args&& ...) [with _Up = std::variant<long unsigned int, double>; _Args = {long long unsigned int}; _Tp = std::variant<long unsigned int, double>; std::allocator_traits<std::allocator<_CharT> >::allocator_type = std::allocator<std::variant<long unsigned int, double> >]’
/usr/include/c++/9/bits/vector.tcc:115:30:   required from ‘std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::emplace_back(_Args&& ...) [with _Args = {long long unsigned int}; _Tp = std::variant<long unsigned int, double>; _Alloc = std::allocator<std::variant<long unsigned int, double> >; std::vector<_Tp, _Alloc>::reference = std::variant<long unsigned int, double>&]’
/home/ubuntu/mdflib/src/ichannelconversion.cpp:318:34:   required from here
/usr/include/c++/9/ext/new_allocator.h:146:4: error: no matching function for call to ‘std::variant<long unsigned int, double>::variant(long long unsigned int)’
  146 |  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
```

ihedvall, 2023-08-03:
No, I have not seen this. We do a Linux build on ubuntu 22.04 (Github Action). This version uses gcc version 11while you using the gcc version 9. The library is fairly new so it's most likely some missing std::variant/std::vector function. 

If you can't update the compiler, there might be a C++20/C++17 issue. GCC 9 support C++17 and not C++20.

Unfortunately, It's hard to regenerate your environment

If it's possible, try to replace in ichannelconversion.cpp on line 318, the emplace_back(..) with a push_back(..) and check if at least that error disapear.

edmendezcr, 2023-08-03:
Thanks! Yes, the issue was the version of GCC. It worked after using GCC 11.

closed 2023-08-03 by ihedvall