ihedvall/mdflib #39: Support for sub-channels or composition of channels

mattbulow-lem, 2023-10-06:
Hi, thanks for your work on this library! 

I am trying to use the mdfWriter for logging raw CAN bus dataframes. I am basing the structure of the MDF file off of the structure from a CSS CANEdge2, which has a structure channel called CAN_DataFrame. This seems to be controlled by ASAM MDF associated standard for Bus Logging.
[image]

I havent been able to figure out how to create sub-channels. Is this feature currently available?

ihedvall, 2023-10-06:
Simply No. The current writer is intended to produce standard measurement files where the signals (channels) belongs to a channel group. The reader handle both CAN log files as well as other types of MDF files. It's a lot of if statements but so far it reads most of the files.

When doing a writer, these if statements must be replaced by user settings which might be a challenge for the user. Instead the plan is to let the user select a type of writer. The current plan is to do a Bus Logger Writer where the user associate a channel group with a message ID and the internal software fix the rest.

There exist a project in the GitHub repository with task/requirements "Developing MDF applications and libraries/Requirement Bus Logger MDF writer.  I will add you as a writer to this project. There exist some ASAM MDF standard how to do a bus logger.

This above might be difficult to understand so if you want we may have a Teams meeting or similar.
Best regards 
Ingemar Hedvall

ihedvall, 2023-10-06:
Just a note on CANEdge
The CSS Electronics (CANEdge loggers) stores the raw CAN messages in a so-called VLSD (Variable Length Signal Data) channel group. The purpose is that the logger can append messages to the end of the file as they come. The disadvantage is that the file is un-finalized, so the reader must finalize the MDF file first and then parse it. Note that not many third-party reader/viewer can handle this.

mattbulow-lem, 2023-10-06:
Some more details of our process
We use the CCS CANEdge as a 'black box' logger that records raw CAN dataframes for vehicles we build. If there is an issue with a vehicle that can be resolved through 'telematics' data then we grab the data from the CANEdge. We need to finalized the files, and modify the CAN_DataFrame.ID channel so it can be read by CANalyzer v11 for anaylsis (Vector tools expect the 31st bit of the ID channel to be populated with an 'extended ID' bit, this is against the MDF standard which is why CSS doesnt do it). Newer versions can read CSS files (CANalyzer V15 and newer). This process can be tedious. So I am working on getting our 'Telematics' unit (linux microcontroller) to be a logger. This unit will not need to create files using VLSD, since it will not experience power loss on vehicle shutdown (internal battery). Our CAN bus does not currently include CAN FD. Not many off the shelf components we deal with support CAN FD for regular communication (some do for flashing FW only). Hope this adds some useful context.

A teams meeting would be great. I am free most days and am located in Colorado, USA. Please send an invite: matthew.bulow@lightningemotors.com

ihedvall, 2023-11-27:
I have added support for composition channels and bus logging.There are some new unit test that show how to use the new features.

closed 2023-11-27 by ihedvall