ihedvall/mdflib #44: how to read CA(Channel array block)?

shaozk, 2023-11-18:
Hi, I want use MDF Viewer to open the mf4 file, but I am unable to display the data in the CA(Channel array block). 

[image]

Is there a method to read data from the CA? I would be very grateful if you could help me.

ihedvall, 2023-11-18:
No, there is no function that reads data from a CA block. The current observers only collects single value/sample through the ReadData() function. 

I need to figure out a solution how to read and store multi-dimensional data. Signal Data (SD) have similar problems. Let me think a day or two. Presentation is also a challenge.

ihedvall, 2023-11-29:
I have started with the work. The CA block is very flexible by its design, structure, arrays and storage. I do have very limited number of example files. 

A CN block may store many values array values/sample. So my proposal is to add an observer interface where the user can fetch not just one value as today, instead fetch all values in the sample. Unfortunately the only solution is that the values are returned in a one-dimensional array. How to arrange them into structures or a multi-dimensional array, is the end-user problem. Is this OK.

When presenting the array as string, it is theory, possible to present the data a more user-friendly way. I had some idea to use JSON format. Do you have any ideas about this?

The third solution is to add some high level presentation in the MDF Viewer through GnuPlot or Python (MatPlotLib) but I need you help how to merge data from several DG/CG blocks. Please give me a hint.

I start to support reading only. Support of CA writing, I take another day.

ihedvall, 2023-12-10:
I have added support for reading of channel arrays. The array values are read as a one-dimensional list as it doesn't exist dynamic multi-dimensional array in C++. 

If you have any idea how to present multi-dimensional arrays in the MDF Viewer application. That application have boost support that do have support for multi-dimensional arrays. JSON strings are also an idea. GnuPlot support up to 4 dimensions arrays but it is some days work to get that working.

closed 2023-12-12 by ihedvall

shaozk, 2023-12-12:
Thank you very much for taking my problem so seriously and quickly resolving it. A one-dimensional list can already meet my requirements. Thank you very much again. You are truly an excellent programmer with profound coding skills.

I have no ideas on how to represent multi-dimensional list, and I am very sorry that I cannot help you. I am a rookie programmer.

ihedvall, 2023-12-12:
I am doing a Release 2.1 now. Will arrive shortly (< 1 h) on GitHub.