ihedvall/mdflib #46: Read byte array error

songjmcn, 2023-11-23:
Hello developers:
  I found a bug when I using this library.
  I use this library to read byte array from MF4 file, this library return error array length. I review the code and find  read the data using wrong function. I change the ichannelobserver.cpp as follow:
[image]
This code can work correct.

ihedvall, 2023-11-23:
Yes you are right. This function are missing a lot of functionality as string to byte array handling (see also the GetChannelValue<string>() function.

To be honest, I never figure out what type of data this MimeStream and MimeSample are. Currently it is treated as string but this can be wrong. Some research might be needed.

ihedvall, 2023-11-24:
There is an underlying problen with the storage of MimeSample and MimeStream channels. I need to fix this. I assign this bug to me and fix it ASAP.

jezzzzza, 2023-11-25:
Hi, i think i have a similar problem. Im trying to save this as byte array in the example (test.cpp)

`      const char* str =
          "testkslflkshd/0fkhslkdfhlksdhflkhsdlkfasdiadslöasjdjäaäsfhpoaihüoahsfd"
          "üoasgüogoüahgpon<psonfd0934htß92hß0h0)?=$?W=&UI&?=AOSNPAsdp0ßj#+#+"
          "üasf";

      channels[5].SetChannelValue(str);

      // Example data to be sent to the channel
      uint8_t data[] = {10, 20, 30, 40, 50, 0, 124, 229, 87};

      std::vector<uint8_t> data2 = {10, 20, 30, 40, 50, 0, 124, 229, 87};

      channels[6].SetChannelValue(data2.data(), data2.size());`
      
      
But in the read section the channel value is empty.

ihedvall, 2023-11-27:
I have started the MIME fix but I can't find any MDF file with MIME sample/stream data. So fixing the bug is difficult to verify. Anybody have a sample MDF file?

ihedvall, 2023-11-28:
I have fixed the MDF library regarding the MIME sample and streams. Please verify with existing file or test.cpp.

jezzzzza, 2023-11-28:
thx for the quick fix.. i tried the above example in test.cpp uint8_t vector but in replay part there is still no data inside.. not sure if i make everything right..

ihedvall, 2023-11-28:
There is a new unit test TestWrite::Mdf4Mime that demonstrate how to use the MIME types. I don't understand the Replay meaning nor have access to the test.cpp file.

jezzzzza, 2023-11-28:
The test.cpp is in mdflibrary_example folder. The SetChannelValue function in test.cpp is a member of MdfChannel class in namespace mdflibrary, this still seems not to work.. the setChannelValue function in testwrite.cpp is a member of IChannel in namespace mdflib and seems to work.

Anyhow, dont know what is the difference between mdflibrary and mdflib, though.. in test.cpp the mdflibrary namespace is used..

ihedvall, 2023-11-28:
MdfLib is a C++ library while MdfLibrary is a .NET assembly. I don't understand why the .NET example is written in C++ instead of in C#.

Simplxss, 2023-11-28:
> The test.cpp is in mdflibrary_example folder. The SetChannelValue function in test.cpp is a member of MdfChannel class in namespace mdflibrary, this still seems not to work.. the setChannelValue [...]

mdflib is a static library. mdflibrary is a shard library that export most functions of mdflib.

Simplxss, 2023-11-28:
mdflibrary_example is not a .NET example. It is a example that how to use the native function in mdflibrary. You could see all functions exported in mdflibrary with DLL Export Viewer.
[image]

jezzzzza, 2023-11-28:
Ok, understood, somehow this function in MdfChannelObserver doesn't return a value, any idea how to fix it?

[image]

[image]

Simplxss, 2023-11-28:
Could you try to use dll form here?
https://github.com/ihedvall/mdflib/actions/runs/7001808212

jezzzzza, 2023-11-28:
not working either unfortunately..

Simplxss, 2023-11-28:
ok, I will fix it soon

jezzzzza, 2023-11-28:
ok thank you!

One more thing..

I had to add this: in order being able to build the whole project, including the shared library..

[image]

that gave me that compiler warnings though.. i ignored them right now..

[image]

  

> [2/42] Building CXX object mdflibrary\CMakeFiles\mdflibrary.dir\src\AssemblyInfo.cpp.obj
> D:\git\mdflib\build\ninja-multi-vcpkg-x64\mdflib\cl : Command line warning D9025: overriding '/EHs' with '/EHa'
>   [3/42] Building CXX object mdflibrary\CMakeFiles\mdflibrary.dir\src\MdfChannelArray.cpp.obj
> D:\git\mdflib\build\ninja-multi-vcpkg-x64\mdflib\cl : Command line warning D9025: overriding '/EHs' with '/EHa'
>   [4/42] Building CXX object mdflibrary\CMakeFiles\mdflibrary.dir\src\MdfDataGroup.cpp.obj
> D:\git\mdflib\build\ninja-multi-vcpkg-x64\mdflib\cl : Command line warning D9025: overriding '/EHs' with '/EHa'
>   [5/42] Building CXX object mdflibrary\CMakeFiles\mdflibrary.dir\src\MdfChannelConversion.cpp.obj
> D:\git\mdflib\build\ninja-multi-vcpkg-x64\mdflib\cl : Command line warning D9025: overriding '/EHs' with '/EHa'

im building with

> Microsoft Visual Studio Professional 2022 (64-bit) - Current
> Version 17.7.6

jezzzzza, 2023-11-28:
Ok, did that..

Now it seems like nth is being copied into value here, value is a nullptr..

[image]

<HTML>
<head>
<body>

  | Name | Value | Type
-- | -- | -- | --
◢ | value | 0x0000000000000000 <NULL> | unsigned char *

</body>
</HTML>

ihedvall, 2023-11-28:
I think the value object should be a reference like the size variable

Simplxss, 2023-11-28:
try to modify `std::vector<uint8_t> value` of `GetChannelValue` to `std::vector<uint8_t>& value`

ihedvall, 2023-11-28:
I'm still lost. If your application is C++, use mdflib. If its a C# application use the .NET assembly mdflibrary.

jezzzzza, 2023-11-28:
@Simplxss not sure what you mean..

@ihedvall I started with mdflibrary example since i thought it is the legit example of how to use mdflib. And it also works fine for single values (float, integer etc...). Later, after i figured that i cannot read back byte arrays i tried to use mdflib but ran into execution errors somehow related to observer_list which is a unique ptr list as far as i remember..

so i thought trying debugging the mdflibrary example might be the faster path..

Anyhow, maybe try rethinking naming here!?

Simplxss, 2023-11-28:
It should be fixed now.

https://github.com/ihedvall/mdflib/actions/runs/7020847467/job/19101525608

ihedvall, 2023-11-28:
If you want a shared mdflib, you may just changes the mdflib add_library(mdf to SHARED... instead of  STATIC or remove it and let the CMake BUILD_SHARED_LIBS=ON(/OFF) option control the build. The latter is the CMake standard approach.

The -NET assembly (mdflibrary.dll) is build by Visual Studio solution file (mdflib.sln). I never got CMake produce .NET application nor assemblies. The .NET assembly is used by some users.

jezzzzza, 2023-11-28:
ok, thank you..

@Simplxss seems like this did the trick.. thank you..

[image]

@ihedvall im building with cmake in vs2022 and now it works fine with the hints from @Simplxss and now it seems to work..

jezzzzza, 2023-11-28:
@ihedvall seems to work only for non-zipped/uncompressed data now.. maybe some more fixes in #51 needed?

ihedvall, 2023-11-29:
The DZ to SD conversion was missing for one type of signal data  configuration. 

There where more error little bit here and there, so I don't know if I can close this bug?