ihedvall/mdflib #48: Create an Python interface

ihedvall, 2023-11-23:
Simply create a Python interface similar to the C# interface.

ihedvall, 2023-12-23:
Looks like pybind11 is a more controlled way of doing the interface. Compared to SWIG.

ihedvall, 2024-01-27:
I have done a pybind11 python interface in the DBC library as a pybind11 test run. A python interface for the MDF library will can now be done fast. 

So far, nobody have required a Python interface for the MDF library, why I put this enhancement into low priority. If anyone needs an interface, please let me know.

CarlesAB, 2024-03-12:
Hello, 

Could you please guide me to the folder where to find this Python pybind11 interface? Is it possible to load the MDF4Reader.dll/MDF4Writer.dll as COM objects? 

Thanks for the help.

ihedvall, 2024-03-12:
The python interface is done if anyone wants it. It should be strait forward but it's still some (2-3) weeks work with the pybind11 interface. I did a training mission  on the DBC library as it is smaller than the MDF library. Do you want me to do the Python interface ?

There is no COM interface for the MDF library but there is a .NET assembly available (mdflibrary.dll). It is build by using the Visual Studio IDE (vcproj/sln). This interface is rarely used and manually maintained. The other libraries use CMAKE. In theory, it is possible to create a COM interface but it's a lot of work to support this type of legacy interface. Excel do still use VB script (COM). Please let me know if a COM interface is a requirement.

CarlesAB, 2024-03-12:
Hello, 

If it’s not too much trouble and if it’s already developed, I would appreciate a Python interface.

I don’t require a COM interface. I’m curious about the location of mdflibrary.dll. If I can load this library in Python using the .dll file, it might be sufficient for me to start building something. If not, C# might be a more convenient choice. Thank you.

ihedvall, 2024-03-12:
OK, I will finalize the Python interface by using pybind11. There is an older 2.1 release in GitHub. The executable is an installation kit that includes the .NET assembly. According to Google it should work to load an assembly into Python but this might be in the twilight zone.

ihedvall, 2025-02-28:
@CarlesAB 
The C++ interface is changing. The python interface is started but not finalized due to lack of resources (time). The current Python interface is "identical" with the C++ interface that is a low-level (flexible) interface. Maybe a simpler high-level interface (minimal) should be good enough. Any requirements?