ihedvall/mdflib #52: Cannot create reference link under cc

TheElesty, 2023-11-27:
I cannot create a reference link under CC to link value to its text representation, that's visible under CN. Is that not implemented or Im just missing something in Docu?

TheElesty, 2023-11-28:
Yes, I'm creating it using MdfWriter, I can see in docu IChanelConversion::Reference(), but I cant use it on MdfChannelConversion.
And yes, it need to use it, to link value - string as enumeration.

ihedvall, 2023-11-28:
OK, you are using the C# interface and yes the Reference interface is missing. I can add it.

TheElesty, 2023-11-28:
oh, so the mdflibrary is c# interface ? im using it in cpp project, and example etc lead to using MdfReader/Writer, so I use that

ihedvall, 2023-11-28:
C++ shall use the mdflib library while the mdflibrary is a C# assembly (C++/CLI). The C# is little bit of a mess as it only works on windows, Well it sort of works on MacOS and did work on Linux, but not any more. There exist both Visual Studio Solution/Project file (mdflib.vcxproj) and CMake config file (CMakeLists.txt).

There is a google unit test directory. Check the testwrite.cpp file that have several example on how to create MDF files. The ASSERT_XX and EXPECT_XXX are Google Unit test macro, so ignore them in your code..

ihedvall, 2023-11-28:
Sorry, the mdflibrary is a .NET assembly not a C# assembly but it should be used in C# applications.

TheElesty, 2023-11-28:
maybe to give you a little bit more info (but still a bit crooped cos of NDA), I'm currently testing if this lib is usable in our env, we use c++ with cmake (in vs2022). currently I'm using the mdflibrary.lib, but from what u wrote i should probably use the mdf.lib, right ? it has both read and write interface ?

ihedvall, 2023-11-28:
Yes, you should use the mdflib only. The mdflibrary is just a .NET wrapper around the C++ library. The C++ library also works on most operating systems/platforms.

TheElesty, 2023-11-28:
ok, and in that case that cc reference will work right ? so this issue can be closed maybe ?

ihedvall, 2023-11-28:
Well, It is some .NET wrapper code still missing so I need to fix that.

The  cropped cos of NDA sounds more interesting?

TheElesty, 2023-11-28:
So Im testing the mdf-lib, I have similar erros as, when i was building mdeflib from source, I got a lot of _LNK2038 mismatch detected for 'ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' errors for debug, and some errors about linking EXPAT (and maybe zlib), I had both expat and zlib installed manually, but probably some wrong versions, cos it missed something from zlib (zlibstatic.lib or similar)

ihedvall, 2023-11-28:
These ITERATOR... is just gibberish messages. Take a step back and check the cmake configuration log output. I print out a lot of trace messages about path for EXPAT and ZLIB.

If you look at CMakeList.txt you can either build using VCPKG or by a Component Directory.

If you intend to use VCPKG, it will build the right EXPAT and ZLIB for you. You need to add the 
-D CMAKE_TOOLCHAIN_FILE = C:/vcpkg/scripts/buildsystems/vcpkg.cmake (or where the VCPKG is installed. GitHub Action uses vcpkg so you can spy on the ".github/workflows/ cmake_windows.yml" file.

The other preferred way is to use component dir.  If you check the script/expat.cmake file, it checks for CMake option COMP_DIR.
If it set to where EXPAT and ZLIB is built it uses that path for ZLIB_ROOT and EXPAT_ROOT. The ihedvall/components repository have all bash/bat scripts needed to create a COMP_DIR. Use CMake -D COMP_DIR = "where you built/installed the third-party library.

The third way is to set the EXPAT_ROOT and ZLIB_ROOT to where you installed the libraries. Use -D EXPAT_ROOT="install path" and ZLIB_ROOT="install path". 

The ITERATOR... message is complaining on mismatch between when linking your application against the MDF C++ lib, so there is something mismatching. Remember to turn the MDF_BUILD_SHARED_LIB=OFF so it doesn't build any .NET stuff. 

But check the cmake log messages

ihedvall, 2023-11-29:
I think I have fixed the Ref Link problems now. There where actual several errors. I have added a C# unit test.

Have you solved the build problem with mixed debug and release libraries/executable?

TheElesty, 2023-11-30:
Hi, I'm trying that c++ lib (prebuild) to run in my test project, but whatever I do I still getting problems from expat (I downloaded it from git), I tried different ways to add it to that project bud unsucsefyly

ihedvall, 2023-11-30:
Don't worry to much, This how C++ projects works. You often get stuck in linker problem and the days fly by. All C++ hackers have done this journey several times. The below CMake log snippet, show my setup with the COMP_DIR approach.
```
-- ZLIB Include Dirs: K:/zlib/master/include
-- ZLIB Libraries: optimizedK:/zlib/master/lib/zlibstatic.libdebugK:/zlib/master/lib/zlibstaticd.lib
-- EXPAT Found: TRUE
-- EXPAT Include Dirs: K:/expat/master/include
-- EXPAT Libraries: optimizedK:/expat/master/lib/libexpat.libdebugK:/expat/master/lib/libexpatd.lib
-- Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES wxWidgets_INCLUDE_DIRS adv core base) 
-- Found wxWidgets: debug;K:/wxwidgets/master/lib/vc_x64_lib/wxmsw33ud_adv.lib;optimized;K:/wxwidgets/maste 
``` 

The GitHub:ihedvall/components repository, contains some scripts. The build_EXPAT scripts are of interest.
We can setup an on-line meeting (MS Teams or similar). My email address is ihedvall@telia.com.

TheElesty, 2023-12-01:
Its probably unnecesery, we will probably use other solution.

closed 2023-12-01 by ihedvall