ihedvall/mdflib #68: undefined reference to `MdfChannelCreateChannelComposition

ali-ghadyali-fvt, 2024-01-09:
I am building from main and configuring using the following command
`cmake -S . -B build/desktop -DCMAKE_BUILD_TYPE=release -DCMAKE_INSTALL_PREFIX=`pwd`/opt/libs/mdflib/desktop -DMDF_BUILD_SHARED_LIB_EXAMPLE=ON -DMDF_BUILD_TOOL=ON -DMDF_BUILD_TEST=ON`

I get a linker error for the above function. The functions seem to be declared using MACROS so it is a bit confusing on where I should be looking for a fix.

ihedvall, 2024-01-10:
I will test your combination but set the MDF_SHARED_LIB=OFF. This disable the .NET mdflibrary build, assuming you not need the C# interface. The MDF_LIB_TOOL=OFF speed up the build time as the wxWidgets third-party library take some time to build. Note that wxWidgets currently doesn't build on Linux/MacOS. It is a VCPKG problem, so there is nothing I can do.

Can you supply me with which operating system you working with?

ihedvall, 2024-01-10:
I tried and I also gets a lot of link errors that indicate that you mixing C++ and C# targets. Setting the MDF_SHARED_LIB=OFF will fix the "mixing" linkage problem but I suspect you will get some other problem with third-party library. If you running on Windows, the VCPKG solves most of the problems with third-party libraries. If you have VCPKG installed, you should add the `-DCMAKE_TOOLCHAIN_FILE=<vcpkg-root>/scripts/buildsystems/vcpkg.cmake`. Expect some extra build time at first build.

ihedvall, 2024-01-10:
If you not using the VCPKG, I recommend to invest in a component directory (-DCOMP_DIR=path to third-party libraries). The ihedvall/components contains script files for most third-party libraries. Little bit more manual work but works on most operating systems.

Simplxss, 2024-01-10:
@ihedvall By the way, is there any example for ChannelCompositions. I will add some test for it.

ihedvall, 2024-01-10:
Channel compositions are typically used for in bus logging. If you want typically bus logger files, the CSS Electronics have some logger files available. There are also some unit tests as in TEST_F(TestWrite, Mdf4MlsdCanConfig).

Hint: If you run the unit test TestWrite, it creates a lot of MDF files in the %TEMP%/test/mdf/write directory. These files will be removed the next test run.

ihedvall, 2024-01-10:
Tested the above cmake setting. The same build issue still exist. Reopen the issue.

ihedvall, 2024-01-10:
I doesn't run without the VCPKG. You can build a shared library on the mdflib by changing the add_library(..STATIC to SHARED and add a set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON). This should eliminate the need of using the .NET interface export.cpp wrapper.

I do get some minor annoyance that I need to copy the dll to the excutable directory after build (as mdfviewer.exe).

ali-ghadyali-fvt, 2024-01-10:
> I tried and I also gets a lot of link errors that indicate that you mixing C++ and C# targets. Setting the MDF_SHARED_LIB=OFF will fix the "mixing" linkage problem but I suspect you will get some [...]

Hi @ihedvall,

I am building on Ubuntu WSL 22.04. I am writing a pure C++ application. I use a custom toolchain file for CMake which includes the paths for wxwidgets. I can get the library to compile and build just fine. I can run the viewer and the library test application included in the project. 

I am using an mf4 file generated from the css canedge for reference to setup my own program. I am logging canbus. I can send you screenshots tomorrow since I am out of office today. The problem occours when I try to use the CreateChannelComposition function. I am unsure of which option is causing just that one function to not being included in compilation. To verify that the problem is within the library itself I tried compiling with tests on and encountered the error again. If you point me in the right direction I will attempt to fix it myself and submit a PR.

ihedvall, 2024-01-10:
I am impressed that you get the test_mdf and mdfviewer running. The boost and wxWidgets are little bit tricky. Both the mdfviewer and the test_mdf executable uses the static mdf.lib instead of the mdflibrary.dll. I recommend that you use the static library in your C++ projects. 

The mdflibrary.dll is designed as a .NET (C++/CLI) interface and is intended to be used in C# projects. You can get  this DLL running on Linux but it require an extra wrapper that is not well tested. In your case it was just an extra character. I think Simplxss have fixed the error, so just a git pull would probably solve the above problem. 

If you run the test_mdf.exe, there should be a number of MDF files in the /temp/test/mdf/write directory. Well until next OS restart at least. 

If you want to read CAN bus traffic, you might be interested in the DBC library instead. The MDF file normally only store the the 8 byte data, Can Id and time. You need a DBC file to convert the data bytes to signals. With the DBC Viewer you may read in a DBC file and the import MDF file afterwards thus converting the bytes to signals and samples. The ihedvall/dbclib is relative new so it might have some unknown features.

ihedvall, 2024-01-10:
If you are writing a CAN bus logger i.e. creating MDF files , you might need some meeting as the writing of MDF files are little bit tricky. You may check the Google Test TestWrite, Mdf4MlsdCanConfig for guidance. Please let me know if you need some meeting.

ali-ghadyali-fvt, 2024-01-11:
> I am impressed that you get the test_mdf and mdfviewer running. The boost and wxWidgets are little bit tricky. Both the mdfviewer and the test_mdf executable uses the static mdf.lib instead of the [...]

@ihedvall The typo correction from @Simplxss fixes the issue. I was able to use the .so in a C++ application. 

I have shown below what the intended file formatting should look like. 

[image]

The image below is of my mdf file using CreateChannelComposition

[image]

The intended use for mdflib for my use case is to bus log files that ASASMMDF can view. Thank you for your offer for meeting.  Is there a work email I can reach you at.?

ihedvall, 2024-01-12:
I have sent my e-mail address through Linkedin.The MdfWriter::CreateBusLogConfiguration() function creates a standard CAN bus configuration for you but you need to add the source information (SI). There are some different ways you can store bus traffic. It might be worth looking at the mdfwriter.h and the canmessage.h. 

Running bus loggers might require some MS Teams meeting.

commit eaa3ed64 (2024-01-10): fix #68
closed 2024-01-18 by ihedvall