ihedvall/mdflib #69: v2.1 build error with vcpkg manifest mode

jezzzzza, 2024-01-23:
Trying to build v2.1 with vcpkg in manifest mode gives following errors.

`
Severity	Code	Description	Project	File	Line	Suppression State
Error		CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Professional/VC/vcpkg/scripts/buildsystems/vcpkg.cmake:893 (message):
  vcpkg install failed.  See logs for more information:
  D:\git\mdflib\build\ninja-multi-vcpkg-x64\vcpkg-manifest-install.log		C:/Program Files/Microsoft Visual Studio/2022/Professional/VC/vcpkg/scripts/buildsystems/vcpkg.cmake	893	
Error		CMake Error at scripts/cmake/vcpkg_download_distfile.cmake:32 (message):
      
      Failed to download file with error: 1
      If you are using a proxy, please check your proxy setting. Possible causes are:
      
      1. You are actually using an HTTP proxy, but setting HTTPS_PROXY variable
         to `https://address:port`. This is not correct, because `https://` prefix
         claims the proxy is an HTTPS proxy, while your proxy (v2ray, shadowsocksr
         , etc..) is an HTTP proxy. Try setting `http://address:port` to both
         HTTP_PROXY and HTTPS_PROXY instead.
      
      2. If you are using Windows, vcpkg will automatically use your Windows IE Proxy Settings
         set by your proxy software. See https://github.com/microsoft/vcpkg-tool/pull/77
         The value set by your proxy might be wrong, or have same `https://` prefix issue.
      
      3. Your proxy's remote server is out of service.
      
      If you've tried directly download the link, and believe this is not a temporary
      download server failure, please submit an issue at https://github.com/Microsoft/vcpkg/issues
      to report this upstream download server failure.		D:\git\mdflib\scripts/cmake/vcpkg_download_distfile.cmake		
Error		CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage				
`

Building the deps of expat and zlib in another vcpkg manifest project builds with no problem.

Really no idea what is wrong, i have deleted my entire vcpkg already and didnt help either..

ihedvall, 2024-01-23:
@jezzzzza 
Can you supply the cmake defines input.  I assume Windows 64-bit build. Example below
 -GNinja
-DCMAKE_TOOLCHAIN_FILE=c:/vcpkg/vcpkg/scripts/buildsystems/vcpkg.cmake
-DCMAKE_INSTALL_PREFIX=o:/mdf
-DMDF_BUILD_TOOL=ON
-DMDF_BUILD_TEST=ON
-DMDF_BUILD_SHARED_LIB=OFF
-DBUILD_SHARED_LIBS=OFF

ihedvall, 2024-01-23:
@jezzzzza 
This is just a guess from me.
The ninja-multi-vcpkg-x64 preset is not found in repository. I think it was renamed to ninja-vcpkg-x64.
If you clear the ../build directory of old contents and change the preset, you might have better luck.

jezzzzza, 2024-01-23:
What do you mean by that.. the CMakePresets.json is part or git project, isnt it? And where can i find the cmake defines input?

jezzzzza, 2024-01-23:
Same error when i try to build on main (after Delete Cache and Rebuild)

`

Fetching registry information from https://github.com/microsoft/vcpkg (HEAD)...
Detecting compiler hash for triplet x64-windows...
The following packages will be built and installed:
    expat:x64-windows -> 2.4.8#1 -- C:\Users\myuser\AppData\Local\vcpkg\registries\git-trees\18754c99766e364a887a08bbf85c7fae06185f36
  * vcpkg-cmake:x64-windows -> 2022-08-18 -- C:\Users\myuser\AppData\Local\vcpkg\registries\git-trees\84c200e8e625d4d99b1649525fcdf81a73197078
  * vcpkg-cmake-config:x64-windows -> 2022-02-06#1 -- C:\Users\myuser\AppData\Local\vcpkg\registries\git-trees\8d54cc4f487d51b655abec5f9c9c3f86ca83311f
    zlib:x64-windows -> 1.2.12#2 -- C:\Users\myuser\AppData\Local\vcpkg\registries\git-trees\d40d86865ecbcc5b54d21f840dd2212556aeadd5
Additional packages (*) will be modified to complete this operation.
Restored 2 package(s) from C:\Users\myuser\AppData\Local\vcpkg\archives in 33.4 ms. Use --debug to see more details.
Installing 1/4 vcpkg-cmake-config:x64-windows...
Elapsed time to handle vcpkg-cmake-config:x64-windows: 4 ms
Installing 2/4 vcpkg-cmake:x64-windows...
Elapsed time to handle vcpkg-cmake:x64-windows: 4.57 ms
Installing 3/4 expat:x64-windows...
Building expat:x64-windows...
-- Installing port from location: C:\Users\myuser\AppData\Local\vcpkg\registries\git-trees\18754c99766e364a887a08bbf85c7fae06185f36
-- Using cached libexpat-libexpat-3bab6c09bbe8bf42d84b81563ddbcf4cca4be838.tar.gz.
-- Extracting source C:/Users/myuser/AppData/Local/vcpkg/downloads/libexpat-libexpat-3bab6c09bbe8bf42d84b81563ddbcf4cca4be838.tar.gz
-- Using source at D:/git/mdflib/build/ninja-vcpkg-x64/vcpkg_installed/vcpkg/blds/expat/src/4cca4be838-7c1b9fd36a.clean
-- Found external ninja('1.11.0').
-- Configuring x64-windows
-- Building x64-windows-dbg
-- Building x64-windows-rel
-- Fixing pkgconfig file: D:/git/mdflib/build/ninja-vcpkg-x64/vcpkg_installed/vcpkg/pkgs/expat_x64-windows/lib/pkgconfig/expat.pc
-- Using cached mingw-w64-i686-pkg-config-0.29.2-3-any.pkg.tar.zst.
-- Downloading https://mirror.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst;https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst;https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst;https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst;https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst;https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst -> mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst...
[DEBUG] To include the environment variables in debug output, pass --debug-env
[DEBUG] Trying to load bundleconfig from C:\Program Files\Microsoft Visual Studio\2022\Professional\VC\vcpkg\vcpkg-bundle.json
[DEBUG] Bundle config: readonly=true, usegitregistry=true, embeddedsha=638b1588be3a265a9c7ad5b212cef72a1cad336a, deployment=VisualStudio, vsversion=17.0
[DEBUG] VS telemetry not opted in, disabling metrics
[DEBUG] Feature flag 'binarycaching' unset
[DEBUG] Feature flag 'compilertracking' unset
[DEBUG] Feature flag 'registries' unset
[DEBUG] Feature flag 'versions' unset
[DEBUG] Feature flag 'dependencygraph' unset
Downloading https://mirror.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst
Downloading https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst
Downloading https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst
Downloading https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst
Downloading https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst
Downloading https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst
error: Failed to download from mirror set
error: https://mirror.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst: failed: status code 404
error: https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst: failed: status code 404
error: https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst: failed: status code 404
error: https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst: failed: status code 404
error: https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst: failed: status code 404
error: https://repo.msys2.org/mingw/i686/mingw-w64-i686-libwinpthread-git-9.0.0.6373.5be8fcd83-1-any.pkg.tar.zst: failed: status code 404
[DEBUG] D:\a\_work\1\s\src\vcpkg\base\downloads.cpp(1003): 
[DEBUG] Time in subprocesses: 0us
[DEBUG] Time in parsing JSON: 16us
[DEBUG] Time in JSON reader: 0us
[DEBUG] Time in filesystem: 4025us
[DEBUG] Time in loading ports: 0us
[DEBUG] Exiting after 654 ms (653264us)

CMake Error at scripts/cmake/vcpkg_download_distfile.cmake:32 (message):
      
      Failed to download file with error: 1
      If you are using a proxy, please check your proxy setting. Possible causes are:
      
[... cut ...]
error: building expat:x64-windows failed with: BUILD_FAILED
Elapsed time to handle expat:x64-windows: 14 s
Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
    https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+expat
You can submit a new issue at:
    https://github.com/microsoft/vcpkg/issues/new?title=[expat]+Build+error&body=Copy+issue+body+from+D%3A%2Fgit%2Fmdflib%2Fbuild%2Fninja-vcpkg-x64%2Fvcpkg_installed%2Fvcpkg%2Fissue_body.md

`

jezzzzza, 2024-01-23:
I have to add a baseline though otherwise i get this error

error: this vcpkg instance requires a manifest with a specified baseline in order to interact with ports. Please add 'builtin-baseline' to the manifest or add a 'vcpkg-configuration.json' that redefines the default registry.

so i added 
  "builtin-baseline": "6ca56aeb457f033d344a7106cb3f9f1abf8f4e98",

like that

{
  "$schema": "https://raw.githubusercontent.com/microsoft/vcpkg-tool/main/docs/vcpkg.schema.json",
  "name": "mdflib",
  "version": "2.1",
  "builtin-baseline": "6ca56aeb457f033d344a7106cb3f9f1abf8f4e98",
  "dependencies": [
    "zlib",
    "expat"
  ],
  "features": {
    "mdfviewer": {
      "description": "Build mdf viewer",
      "dependencies": [
        "boost-process",
        "boost-locale",
        "wxwidgets"
      ]
    },
    "mdflibtest": {
      "description": "Build mdf tests",
      "dependencies": [
        "boost-endian",
        "gtest"
      ]
    }
  }
}

ihedvall, 2024-01-23:
@jezzzzza 
Earlier, I got problem with the MSYS2 and MSVC installed on the same machine. CMAKE find some libraries in the MSYS2 world and some in the MSVC world. 

The "builtin-baseline" is a new feature which seems to refer to VCPKG GIT version. I try to update the vcpkg to latest and check if the errors pop-ups.

jezzzzza, 2024-01-23:
OK, thanks for the reply! What is MSYS2?

jezzzzza, 2024-01-23:
Btw. i have only in mdflib project the mentioned problem.. in other projects when i add expat and zlib as deps, it installs fine..

jezzzzza, 2024-01-23:
This command just adds   "builtin-baseline": "2b681d95e08447fde261b4404fbeab324f7d1e5b" to the vcpkg.json.. same error remains..

jezzzzza, 2024-01-23:
We tested also on another PC of a colleague, and it is the same issue there as well..

jezzzzza, 2024-01-23:
I think i found the problem..

I had to change VCPKG_ROOT env variable in presets json to VCPKG_PATH.. VCPKG_ROOT takes some vcpkg install from Visual Studio and not the git clone..

closed 2024-01-23 by jezzzzza