ihedvall/mdflib #73: Setting precision in conversion block does not set precision flag

afestini, 2024-02-15:
https://github.com/ihedvall/mdflib/blob/03586d71769803c6c732af152aa00feb4be7db1b/mdflib/src/cc4block.cpp#L103

Should this be setting CcFlag::PrecisionValid ?

ihedvall, 2024-02-15:
Setting the precision flag and improved unit  test to detect the error.

closed 2024-02-15 by ihedvall

ihedvall, 2024-02-15:
@afestini 
The MDF library have a Reader and a Writer interface. Maybe it should be a Converter interface as well. Please let me know, if you require any improvement on the interface.

afestini, 2024-02-15:
Not quite sure in which way you mean it as interface. I just noticed the channel block was setting the flag when configuring the decimals, so I wondered if the conversion block shouldn't be consistent. Having them set automatically seems very convenient (and you can still override the flags otherwise).

ihedvall, 2024-02-16:
@afestini 
Sorry about the misunderstanding. You can skip my previous message. It has nothing to do with the CC block interface. It was more of a general question about the MdfWriter interface.