ihedvall/mdflib #82: .dll for direct usage inside .net 6

kct-cms, 2024-05-24:
Hello,

thank you for developing a MIT version of the mdf4 file format without any usage of python.
I'm currently struggling to get all dependencies installed and ready to build a .dll for my C# project.
Is there any way to create a release version of the builded .dll you are currently using in your .net testing classes?

Kind Regards
Timo

ihedvall, 2024-05-24:
The C++/CLI wrapper around the mdflib C++ library is named mdflibrary. It is preferable built using the mdflibrary.vcxproj and mdflib.vcxproj (Visual Studio) files instead of CMAKE. There is a mdflib.sln VS solution file that have some C# unit tests. I suspect that it uses dotnet 8 so this have to be changed.

Note that the Visual Studio environment is not always up to date with the CMAKE changes. Dependencies in C++ are little bit tricky. VCPKG seems to solve most of that problem, well at least in Windows.

I'm not sure if the above solves your problem.

ihedvall, 2024-05-28:
@kct-cms 
Hello Timo
I'm  not sure that my previous message was of any help or it just confusing you. Please let me know.
Best Regards
Ingemar Hedvall

kct-cms, 2024-06-03:
Hello @ihedvall 
thank you for your prompt reply. Sorry for answering this late I was on holiday last week.
I tried building the solution with a .NET 8 environment but expat.h and zlib.h is currently missing.

I tried vcpkg and it looks way better than before but I need a firewall rule to download the zlib package over https.
I contacted my IT department and hopefully, I can retry it later today,

Thank you so far 👍 .

I have a question regarding usage.

Do you have experience using the built .dll with the included C++ libraries? I would like to use the .dll later without installing the required C++ libraries. Are they fully contained in the .dll or is an installation of the dependencies always required?

Best regards,
Timo Kovacevic

ihedvall, 2024-06-03:
The mdflib project is a C++ library that you can link as DLL or as a static library. The latter is normally preferred. The static linkage result in larger executable but there is no need of loading external dependencies as expat.dll and zlib.dll. 

The mdflibrary project generate a .NET assembly. The assembly is always linked as a DLL. The assembly uses a special C++/CLI compile flag and is just a wrapper around the C++ mdflib project. C# project can now use this assembly (DLL) unknowing that is actual a C++ library.

You should link the mdflib statically and include that static library in the mdflibrary assembly. The DLL will be larger but no dependencies. You must however build the expat (XML) and zlib (compress) project as static libraries. This is the VCPKG job. The VCPKG is unfortunately little buggy and risk regarding getting viruses so your IT department might have some arguments here.

The VCPKG is a fairly new so I uses the older technique of doing the VCPKG job manually. It's requires that you build the expat and zlib libraries yourself. This a so called component disc/directory. I have made build scripts for Windows and Unix in the ihedvall/component repository.  You need to modify the mdflib project so it now include path to your component disk/directory root. In CMAKE the COMP_DIR is set with the root directory. The VS project files should have something similar but if you have your own unique vcxproj files is not the end of the world.

kct-cms, 2024-06-03:
Hi,

after adding the firewall rule, VCPKG could download and install the required zlib library.
Now I can build and run the .NET 8 test solution.

Thank you for your help.

Kind Regards,
Timo

kct-cms, 2024-07-23:
Hi,

Sorry for reopening this issue, but I'm having problems creating a clean standalone .dll. I'm using the mdflibrary.dll inside my C# project, and for some reason, my colleague is getting the following exception code:

```
The active test run was aborted. Reason: Test host process crashed: Fatal error. System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at MdfLibrary.MdfWriter.StartMeasurement(UInt64)
   at His.Ocean.Mdf.MdfWriter.Write(His.Ocean.Mdf.MdfFile, System.String)
   at His.Ocean.Skipper.Test.Mdf.MdfTest.WriteMdf_ValidMdfDataStructure_ShouldWriteMdfToFile()
...
```

His local PC has no kind of C++ environment, while mine does since I compiled the .dll. I can run the same unit test as he does, but mine is creating the file without any problems.

I tried creating the .dll with different setups, using the mdflib as a static library, but I guess I might need the expat and zlib as static libraries as you already mentioned in your previous comment. I looked at your repo ihedvall/component and, unfortunately, I'm having problems setting everything up. Due to our current IT policy, it's a bit tedious to get everything together.

So my question is: could you provide me with the static libraries or the mdflibrary.dll directly? I'm really sorry since you already gave me a lot of information and comprehensive libraries to work with, but with my previously non-existent experience in C++ and their library handling, I'm currently at a standstill. Especially since it's running on my PC but not on his, the testing part is really time-consuming.

It would be greatly appreciated if you could help me in this regard. Otherwise, feel free to close the issue again.

Thank you!

ihedvall, 2024-07-23:
The "official" way is to set a release (2.2). I can fix so the .NET dll (Windows 11) is compiled. It will compiled against the latest .NET version. You might also need to install the latest MS VC runtime dll. The both normally comes with the Visual Studio installation as you seems running a C# project. You need to copy the dll from the GitHub.

The C# interface is not 100% maintained by me as it doesn't use the CMAKE build process. Instead it uses the Visual Studio build process.

Your crash might be just a bug depending on where the MDF file is written. There is a known problem to create an MDF file directly on a network disc (NAS), so create on local disc and then copy it. It's something with the NAS write cache/optimization that causes the problem (Write Thru Flag = false). 

Note that most unit tests fails due to that the test MDF files are missing. I have cracked how to conditionally skip a test. Need some C# experienced guy for the maintenance. 

It take some day(s) to do the release as it is some new stuff that also needs to be included (mdf2csv application).

kct-cms, 2024-07-24:
@ihedvall 
Hello Ingemar,

thank you for your fast reply and support. I will check on the unit test again and see if there's any MDF file missing. It would be weird for me since the exact same test, in the same repository, worked on my PC, but you never know. 
The test file is not on a NAS disc, since all our projects are on local disc inside a repository.

A "official" 2.2 release with a included .dll would be amazing and a great benefit. 

We will update our .NET 6 project to .NET 8 next week, the latest .NET version would therefore be great for us.
With C# interface do you mean the C++ Wrapper application "mdflibrary" that is compiled to a .dll?
We are currently running on a x64 Win10 machine, I guess Win11 should be no problem or is this a criteria for C++ Code?  

Thank you for your help!

ihedvall, 2024-07-28:
I have done the MDF Library version 2.2 release. You can download a pre-compiled mdflibrary.dll among the release files.

ihedvall, 2024-08-08:
@kct-cms 
Did the pre-compiled assembly DLL work for you?

kct-cms, 2024-08-08:
@ihedvall 

Hi Ingemar,

thank you, it worked great!
We can now create MDF files without any system setup or manual installation. My team and I are very grateful for your help.

Personally, I would love to send a small tip as a token of appreciation, so you could buy yourself a drink. I couldn’t find any donation link associated with this GitHub project, do you have one?

Kind regards,
Timo

ihedvall, 2024-08-08:
@kct-cms 
No, I don't have sponsor/donation link. The MDF library was original mostly used inside my C++ projects but seems to be used in more or less commercial applications. Maybe it's time to look for sponsoring or similar. I need to solve the Long Time Support issue sooner or later.

Instead of money, I'm more interested about spin-off project ideas. I do need some directions from users here. 

I close this issue for now. Next release will be 3.0 which basically includes the Python support (pybind11 wrapper).  The C++ library have some features that the dotNet assembly are missing. It mainly regarding channel arrays and reading large MDF files (partial reading).  Let me know if you are needing these features.

closed 2024-08-08 by ihedvall