ihedvall/mdflib #84: Linking attachments in case of buslogging

SebastianCML, 2024-05-29:
Hello,

as the specification says, attachments can be linked to channels. This is especially for buslogging from interest, since it allows to attach a .dbc attachment directly to the buslogging channel. Tools like AsamMDF then directly interpret the buslogging data and show the signals.

From the documentation I can read, that the 'Sync' channel is used to synchronize an attachment block. Can you give me an example for the implementation?

Greetings,
Sebastian

ihedvall, 2024-05-29:
The MDF library is "independent" of other libraries. The ihedvall/dbclib have the parser functionality of a DBC file. This library also have interface so you can load in a bus logger file into the parser and get the signal values. The DBC library has similar subscriber interface as the MDF library. In the DbcViewer application, you can load in a MDF bus logger file and plot the signal values.

There is some problem with the parsing of CAN messages regarding type of protocol J1939, NMEA 2000 etc. I think I manage to test the NMEA 2000 and plain CAN protocol but the remaining protocols are untested. So if you have a MDF bus log file with matching DBC file, you are welcome to test or send me the files.

Note that the MDF file seldom attach the DBC or A2L file but if you add an ODS database (ihedvall/odslib) for indexing the files, solves that problem. 

Regarding the Sync channel type, I am at your level, I don't really know how it works. I assume that the attachment is a video stream for example recording something during the test. The sync channel the reference the streams "timestamp" whatever this is. I actually have an synchronization MDF file so I can research little bit more if you want.    

Best Regards
Ingemar Hedvall

ihedvall, 2024-05-29:
I checked how a Sync type channel works. The attachment is added as usual and I assume that it is a video stream of some kind. The Sync channel is similar to the master channel. The master channel defines the channel groups synchronization domain. Let use time in this example. The Sync channel defines the synchronization domain of the video stream which refers to the relative start time of the video.

The channel is defined as a Sync Type. Its signal data points to the attachment block and has its synchronization domain set to time.

There is some "chicken or egg" problem with this design as you defines the Sync channel first, then add the signal data and at the end attach the video stream. When adding the attachment at the end, the synchronization channels signal data link needs to be updated. I'm not sure the current mdflib interface support this when creating MDF files. Maybe need a extra optional channel argument when adding the attachment. Adding this to the to do list.

SebastianCML, 2024-05-29:
Hello,

thank you for the fast answers. to clarify what i exactly mean: 
In case of Buslogging it is genrally possible to attach a .dbc file to a Channel. In MDF FileViewer, it looks like this:
[image]

As an effect, buslogging mdf files are processed by tools like AsamMdf (V 6.4.4) and look like this:

[image]

So, as you can see, pure buslogging data are extracted directly and shown as separate channel groups. This is possible because of a link block between channel and attachment. 

As you already mentioned, the channel is created at the beginning of the measurement, while the attachment is added at the end. But, if the channel block is already written to the file, it is not possible to add the missing linking block right?

Greetings
Sebastian

ihedvall, 2024-05-29:
The Python AsamMdf application have more features than the MdfViewer application. The MDF Viewer is just a plain file browser and doesn't include any knowledge about DBC files. It is not a replacement of the AsamMdf GUI. 

The DbcViewer application have something similar to the AsamMdf GUI application regarding DBC and MDF files. 

The ReportExplorer application have an ODS database (Sqlite) for indexing the test object data so it can handle up to 10 000 files. It will do what you want but it is under development.

If you want me to add this functionality to the MdfViewer, you may request this function. The best way is to add a requirement to the GitHub project (ihedvall/projects/Developing MDF applications and libraries). 

Note that the Sync channel issue is something unrelated. It is when a video stream is added to the MDF file. Maybe it is better to separate that problem for the moment.

ihedvall, 2024-06-18:
I will modify the add attachment function so it fixes the reference link in channels.

ihedvall, 2024-06-25:
Fixed so it is possible to reference attachments in channels. 

Making the same functionality of the AsamMdf GUI tool regarding embedded DBC files is possible but will take some time. A proposal is to open the DBC Viewer application with the bus logger MDF and DBC files as input arguments. This functionality already exist today but through menu selection.

I change this issue to an enhancement.

ihedvall, 2024-06-25:
By the way, you must add the DBC attachment before calling the InitMeasurement() when creating MDF files. I note that the attachment links in the CN block are dynamically i.e. the size of the block changes. You cannot change the block size after it is stored. You may change the content.