ihedvall/mdflib #100: MDF File Metadata Issues

simonkimi, 2024-08-14:
Hello! 

I am using the .NET MDF library to generate MDF files, but I am encountering some issues when inserting metadata.

My code is as follows. When AddCommonProperty is commented out, the generated file works fine. The context is as follows:
```csharp
using System;
using MdfLibrary;

var writer = new MdfWriter(MdfWriterType.Mdf4Basic);
writer.Init("./demo.mdf");

var header = writer.Header;
header.StartTime = (ulong)(DateTimeOffset.Now.ToUnixTimeMilliseconds() * 1000000);

header.Author = "DemoAuthor";

// header.MetaData.AddCommonProperty(new MdfETag
// {
//     Name = "Test",
//     ValueAsString = "Test1",
//     Type = "string"
// });

var meta = header.MetaData;

writer.InitMeasurement();
writer.FinalizeMeasurement();
writer.Dispose();
```

[image]

[image]

However, after uncommenting the AddCommonProperty part, an XML header was added to the XmlSnippet, which caused both the [Vector MDF Validator](https://www.vector.com/int/en/products/application-areas/ecu-calibration/measurement/mdf/) and [asammdf](https://asammdf.readthedocs.io/en/latest/index.html) to fail to recognize it.

[image]

[image]
English: Invalid XML: Invalid XML declaration

[image]

After reviewing the code, I found that skip_header in [IXmlFile::WriteString](https://github.com/ihedvall/mdflib/blob/20c135d82ede75805fe7883eb7ef4091c456ebff/mdflib/src/imetadata.cpp#L160) was set to false. I am not sure whether this is a bug, a feature of MDF, or if there is an issue with how I am using it.

Another issue: I want to fully customize my MetaData, but before I set any values like `header.Author`,`header.MetaData` is null. I don't know how to initialize it.

Looking forward to your guidance.

ihedvall, 2024-08-14:
I am computers are offline for some days but it have been issues with this earlier. I will be online again in the weekend.

default.png (view on web)
default.png (view on web)
default.png (view on web)
default.png (view on web)
default.png (view on web)

ihedvall, 2024-08-14:
I am offline for some days but it has been problem with this before. I will back in the weekend.

simonkimi, 2024-08-15:
No worries at all. I’ve temporarily resolved the issue by creating a fork. We can discuss it further when you’re ready.

ihedvall, 2024-08-18:
I have corrected the bug in the C++ library. Two calls to the WriteString(true) was missing the argument true. I suppose this will solve the initial problem. The C++

I normally build the .NET C++/CLI wrapper with the Visual Studio solution (mdflibrary.sln/vcxproj).  I will check the .NET interface tomorrow as it was original designed for read-only support.

ihedvall, 2024-08-19:
I have added a MdfHeader::CreateMetaData() function that either return the existing object or creates a new one. I have not checked but .NET should have some XML library that could generate XML string in/out without a header. 

However there are a lot of improvements to add to the .NET interface regarding the meta data. The C++ library have some more features that the .NET wrapper is missing.

closed 2024-08-28 by simonkimi