ihedvall/mdflib #106: .Net lib

Mr-ZZH, 2024-11-11:
Lib on the. net platform only has a 64 bit version. Can you share the 32-bit version or open source the code.

ihedvall, 2024-11-11:
@Mr-ZZH 
This might be an easy job or a lengthy job. All the MDF Library code is open source. The C++ library uses CMake for the build while the .NET Library use the Visual Studio (*.vcxproj, *sln). There is a VS Solution file in the repository root (mdflib.sln) that is used for building the mdflibrary.dll assembly. It uses VCPKG for building 3-party libraries. The simplest solution is add a 32-bit build type into the projects and hope that everything build.

Note there is a disadvantage of using 32-bit as it only can allocate <2 GB  of memory. Large MDF file (>1G) may cause a problem. I recommend to upgrade the 32-bit system/executable to 64-bit.

Sometimes the build of the .NET project isn't synchronized with the CMake on main but the version. 

You can try to build yourself or I can make a 32-bit build, but it requires a version change which is a days of work. 

Best Regards 
Ingemar Hedvall

Mr-ZZH, 2024-11-11:
@ihedvall 
Thank you for your reply. Due to historical reasons, I had to use a 32-bit program. And I can't find the code to build the C # version in the repository

ihedvall, 2024-11-11:
@Mr-ZZH 
You can open the mdflib.sln file in the mdflib repository root by using the Visual Studio 2022. There is a historical mistake that the C++ library is called mdflib(.dll) while the .NET assembly is named mdflibrary(.dll). This has caused problem earlier.

I did build a 32-bit version. I disabled the pre-compiled header and added 2 defines., _CRT_SECURE_NO_WARNINGS and _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS. This was done in the mdflib.vcxproj C++ project.

There is some warnings about size_t vs. uint64_t. There where some similar complaints in the C# test code about the ulong vs. uint. 

Besides the warnings, the C# unit test seems to be working.

ihedvall, 2024-11-12:
@Mr-ZZH 
Please let me know if you want me to do a new version with a 32-bit .NET assembly included.

Mr-ZZH, 2024-11-12:
@ihedvall 
If it's convenient for you, that would be great. Thank you!

ihedvall, 2024-11-14:
@Mr-ZZH 
Sorry about the delay.
I have made a release 2.3 of the MDF library. The compiled versions of the .NET mdflibrary.dll 32/64 bit versions are in a mdflibrary.zip file within the release file.

ihedvall, 2024-11-14:
@Mr-ZZH 
Please check if the 32-bit .NET dll work in your environment as I have very limited way of testing on 32-bit systems.

Mr-ZZH, 2024-11-18:
@ihedvall 
Thank you, I tried the new version of the library and it worked fine on my 32-bit program

closed 2024-11-18 by ihedvall