ihedvall/mdflib #115: Need some help with library (using with C#)

sensboston, 2025-02-03:
Hi, I'm working on app for ECU (using CAN bus), and need to implement data backup/restore to mf4 file feature. Fortunately I've found this library - great work, thank you! 

But I have questions about proper library usage; unfortunately, I am not an expert in this area, and the specifications are quite extensive and complex.

<details>
  <summary>So here is my code snippet used to save data (when acquisition completed and values stored to the RAM, I don't need to store 'em "on the fly")</summary>

```
        public void SaveMF4File(string filePath, List<XCPSignal> signals)
        {
            // First, delete existing file if any
            if (File.Exists(filePath)) try { File.Delete(filePath); } catch { }

            // Create MdfWriter
            var writer = new MdfWriter(MdfWriterType.MdfBusLogger);
            writer.BusType = MdfBusType.CAN;
            bool initOk = writer.Init(filePath);

            if (initOk)
            {
                // Configure MDF header
                var header = writer.Header;
                header.Author = "CSi-PROM User";
                header.Department = string.Empty;
                header.Description = string.Empty;
                header.Project = "CSi-PROM";
                header.Subject = string.Empty;

                foreach (var signal in signals)
                {
                    var dataGroup = writer.CreateDataGroup();
                    var channelGroup = dataGroup.CreateChannelGroup();

                    // Set ChannelGroup.Name by signal name:
                    // I can't find grouping information at DBC or A2L files, sorry
                    channelGroup.Name = signal.Name;

                    // Create group for signal
                    //
                    // First, a timestamp channel (double => 8 bytes, "master" time)
                    var timeChannel = channelGroup.CreateChannel();

                    timeChannel.Name = "t";
                    timeChannel.Unit = "s";
                    timeChannel.Type = ChannelType.Master;
                    timeChannel.Sync = ChannelSyncType.Time;
                    timeChannel.DataType = ChannelDataType.FloatLe; // 8 bytes for double
                    timeChannel.DataBytes = 8;

                    // Second, a signal value channel (float => 4 bytes)
                    var valChannel = channelGroup.CreateChannel();
                    valChannel.Name = signal.Name;
                    valChannel.Description = $"Float value for '{signal.Name}'";
                    valChannel.Type = ChannelType.FixedLength;
                    valChannel.DataType = ChannelDataType.FloatLe;
                    valChannel.DataBytes = 4;
                    valChannel.Unit = signal.Units ?? "";
                    valChannel.Range = new Tuple<double, double>(signal.Min, signal.Max);

                    // Start 'measurement' i.e. storing the data
                    // Note: we're storing data not during acquisition but after
                    writer.InitMeasurement();
                    var tickTime = signal.StartTimestamp * 1000000;
                    writer.StartMeasurement(tickTime);

                    // Write all samples for each signal
                    var createdGroups = dataGroup.ChannelGroups;

                    // Store timestamps and values
                    foreach (var meas in signal.Values)
                    {
                        // Store the double in the time channel:
                        ulong timeDouble = (ulong)meas.Timestamp;

                        // Set the channel values
                        timeChannel.SetChannelValue(timeDouble);  // double
                        valChannel.SetChannelValue(meas.Value);   // float

                        // Save the sample in this channel group
                        writer.SaveSample(channelGroup, tickTime);
                        tickTime += (ulong)signal.DAQInterval * 1000000;
                    }
                    // Stop & finalize
                    writer.StopMeasurement(tickTime);
                    writer.FinalizeMeasurement();

                }
            }
            else throw new Exception($"Failed to initialize MdfWriter with file '{filePath}'.");
        }

```
</details>

Although the resulting mf4 file is not completely "correct" and many properties aren't filled in, at least it can be successfully opened by MDFViewr, asammdf and CANopy. 

Could you please provide me a small code snippet to read mf4 file back? 

Also, I do have some additional questions:
- mf4 file saved by CANopy has measurements (signals) grouped by "meaning" i.e. signals has group names like "Engine_Parameters_1", "Engine_Parameters_2", "Engine_Faults", "FC_Commands". I'm using A2L and DBC files to construct my acquisition list of channels/signals but can't find a proper mapping in these files. Do you know where is CANopy saves this information? Currently I decided to use signal names.
- I would like to save some additional information for the each channel/signal in the mf4 file (to use on data restore and reconstructing channels) like a ``` public TXCPHandle Address { get; set; }``` (used for acquisition via XCP), how can I do it? 
- could you also tell me, which properties I should set for mf4 file header/signals/metadata etc. I'm interesting in "must have" properties/data only for now. 

Thanks a lot!

ihedvall, 2025-02-03:
@sensboston 
The MdfLibrary is a C# wrapper around the MdfLib C++ library and yes, the naming is not the best. The C# wrapper was made specific to an end-user why the user documentation mostly refers to the [C++ documentation](https://ihedvall.github.io/mdflib/).

Normally the CMAKE is used to build the library but the C# assembly used the Visual Studio editor instead (mdflib.sln). The solution file includes 3 project. The C++ library, the C# (C++/CLI) assembly and a unit test application. You may check out the unit tests as they mostly write a file and then reads the same. There are much more unit tests in the C++ library. Note that the VS solution file sometimes isn't up to sync with the CMAKE project.

Lets start with the MDF writer. There is no error in your code but the writer is normal storing bus messages or data values (signal values). When creating an MDF file, there is a header block and at least one file history block. The header block should in a perfect world, hold all so-called meta data i.e. information about the test and test object. 

All data is stored in one Data Group (DG) block. Inside the DG block one or more Channel Group (CG) block. Each CG block consist of a so-called master channel which mostly is the sample time (relative time to the start time in the header) followed by either one
or more channels (CN) blocks. The CN block may have a Channel Conversion (CC) block that converts the channel data to a scaled engineering.

Just to confuse you more. A bus logger normally just store the time stamp, message ID and the message bytes (storing some flags as well). Converting message bytes to signals are done outside the MDF file. When reading this file back, you basically just get the raw message bytes. There is a naming standard for bus logging.

Below is a test case for a bus logger (both writer and reader).
```C#
 
public void TestBusLogging()
{           
    const string testFile = "can_bus_logger.mf4";
    if (File.Exists(testFile))
    {
        File.Delete(testFile);
    }

    {
        var writer = new MdfWriter(MdfWriterType.MdfBusLogger);
        Assert.IsNotNull(writer);
        Assert.IsTrue(writer.Init(testFile));
        writer.BusType = MdfBusType.CAN;
        writer.StorageType = MdfStorageType.VlsdStorage;
        writer.MaxLength = 8;
        
        var header = writer.Header;
        header.Author = "Caller";
        header.Department = "Home Alone";
        header.Description = "Testing i";
        header.Project = "Mdf3WriteHD";
        header.StartTime = (ulong)(DateTimeOffset.Now.ToUnixTimeMilliseconds() * 1000000);
        header.Subject = "PXY";

        var history = header.CreateFileHistory(); 
        history.Time = (ulong)(DateTimeOffset.Now.ToUnixTimeMilliseconds() * 1000000);
        history.Description = "Initial stuff";
        history.ToolName = "Unit Test";
        history.ToolVendor = "ACME";
        history.ToolVersion = "2.3";
        history.UserName = "Ducky";

        Assert.IsTrue(writer.CreateBusLogConfiguration());
        var lastDg = header.LastDataGroup;
        Assert.IsNotNull(lastDg);

        var dataGroup = lastDg.GetChannelGroup("CAN_DataFrame");
        Assert.IsNotNull(dataGroup);
     
        Assert.IsTrue(writer.InitMeasurement());
        var timestamp = (ulong)(DateTimeOffset.Now.ToUnixTimeMilliseconds() * 1000000);
        writer.StartMeasurement(timestamp);
        Console.WriteLine("Start Time: {0}/{1} ", writer.StartTime, timestamp);
        Assert.AreEqual(writer.StartTime, timestamp);

        var message = new CanMessage();
        message.MessageId = 12;

        message.DataBytes = new byte[] {0x01,0x02};
        for (var sample = 0; sample < 100; ++sample)
        {
            writer.SaveCanMessage(dataGroup, timestamp, message);
            timestamp += 10000000;
        }
        writer.StopMeasurement(timestamp);
[... 34 lines cut ...]
            var dataValue = Array.Empty<byte>();
            var dataValid = data.GetEngValueAsArray(sample, ref dataValue);            
            var dataString = "";
            data.GetEngValueAsString(sample, ref dataString);
            Console.WriteLine("Valid: {0} {1} {2} {3}", sample, timeValid, identValid, dataValid);
            Console.WriteLine("Sample: {0} {1} {2} {3}", sample, timeValue, identValue, dataString);
            Console.Write("Sample: {0} ", sample);
            foreach (var val in dataValue)
            {
                Console.Write("{0:X2} ", val);
            }

            Console.WriteLine("");
        }
    }
}

```
I noticed that you want to store signal values instead of message bytes. In your writer code, you iterate over the signals but you should iterate over the message instead. DG = measurement, CG = message and CN/CC = signal/channel.  For each message, you need to translate the message bytes to signal values and save these values and the call the SaveSample() function. Note that this method is very as you basically have to convert the signal values back to raw data bytes in the file.

The problem with the above method is that it assumes that one sample is equal with to one message. CAN is however not so funny as it is limited to 8 bytes so signal values that contains more than 8 bytes (diagnostic), depends on a CAN transport protocol. 

Note that I have both DBC and A2L libraries available so it is possible to find another solution to your problem.

To summarize, first choose to store bus logger bytes or signal values. If signal values, you need to iterate over XCP messages rather than XCP signals. Note that the CN object holds the last reported value (LRV).  One XCP message is one sample.

ihedvall, 2025-02-04:
@sensboston 
Sorry, I missed that you didn't run the writer "on the fly". On the fly  is the normal MDF writer application and you append new samples at the end of the file. This means that all values/samples are stored i the same data group (DG). When reading the file, you need to read in all bytes which is slow.

The MDF file can be sorted. This means that you split the above DG/CGs into several DG blocks. Each DG block only contains one CG block. A CG block consist of one or more signals with the same time stamp and number of samples. In your application, you should group the XCP signals into there XCP messages. In your code you have 2 signals (time+ value) per CG block which is OK but inefficient.

The A2L file describes the memory layout of the ECU while the DBC file describes the XCP messages. The A2L file have ludicrous number of signals why many tools generate a so-called MAP file which is a selection list of A2L names. The ECU have normally 2 timers and the XCP master set up a subscription on XCP messages. These message are stored in the DBC file. 

The XCP configuration tool configure the XCP messages with a message name as "Engine_Parameters_1". The DBC file holds the message group name and what signals it transfer. The DBC file also store the bit length, bit position and the bit order. It also stores the scaling factor but only linear scaling.

The A2L file stores the scaling in the conversion methods (CHARACTERISTIC). Not all conversion are linear as in the DBC file. The DBC file only store the signal name. A signals meta data should be fetched from the A2L file. So each CG blocks hold one or more CN/CC block. The CN block hold the basic signal description and its unit while the CC block holds the conversion method (CHARACTERISTIC). 

The DBC file message name is most likely used to group signals together. The "Engine_Parameters_1" name is most likely the DBC message name. The CANape tool may however use some internal file. 

Additional information is normally called meta data (MD) block. Most MDF block can store meta data as an XML string. For CG and CN blocks, you can also store similar meta data in a specific Source Information (SI) block. The latter is stricter and I assume easier for the reader tool to present. The MDF library is somewhat weak regarding meta data (MD) block handling. The C++ library have its own XML parser which it uses but this parser is not available in the C# wrapper. See the Vector bus logger file below.

There is no requirement for but there is some recommendation how the XML should look like. It's 2+ pages of information in the specification. All the information (SYMBOL_LINK) comes from the A2L file. Maybe it is better to have some special adapter objects for this type of information. I don't have fast fix for this.

Below is an example from vector bus logger. Note that the logger actual on stores the raw XCP message bytes + time stamp. It have added so called composite channels to the message byte array. You can both fetch signal and message data. It is little bit more configuration for loops.

You should note that MDF files are designed for on-fly appending. They are slow to store data as well as to get data. Traditional the MDF file is converted into Parquet file(s). This is the Data Lake approach.
 
[image]

sensboston, 2025-02-06:
@ihedvall,
**Ingemar**, sorry for the late reply and thank you for explanation and suggestions! As of today, I have achieved planned functionality for exporting and importing mf4 files, I added some extra data needed for software as metadata, also tried saving data during acquisition, and that works fine too. In the future, I will no doubt to expand functionality and I hope you can help with advice if I have questions.

P.S. I hope you'll add a PayPal or BuyMeACoffee donation links into your repositories? I (and I think not only me) would be happy to show my appreciation in this way 😉 

P.P.S. There is one more... not an issue or even a feature request but kinda... But I'll open a new issue, OK?

closed 2025-02-06 by sensboston