ihedvall/mdflib #116: Embedding CLR library version by Costura/Fody

sensboston, 2025-02-06:
First, it's definitely **not** a bug or even not a feature request. I'm big fan of portable apps and always trying to reduce number of files required for installation (ideally it should be only one executable, with generated settings file in readable and simple xml format, located at the app's folder)
I'm using (in the most cases) [Costura/Fody](https://github.com/Fody/Costura) package: it's open source, MIT licensed and doing job perfectly fine. 

However I do have a pretty strange issue with CLR version of mdflibrary: sometimes calling mdflibrary for the first time raises an exception _"Attempt to load an unverifiable executable with fixups (IAT with more than 2 sections or a TLS section.) (Exception from HRESULT: 0x80131019)"_

This happened not all the time: if CLR assembly already loaded, it's continue working fine. Now, as a workaround, I've excluded mdflibrary.dll from the files processed by Costura, added as an embedded resource, extracted data to dll located in the AppDomain.CurrentDomain.BaseDirectory (otherwise C# can't load assembly because of security reasons) and load assembly manually by calling Assembly.LoadFrom(dllPath); 

So my question is: how to properly embed CLR version of library to the C# solution/project? BTW, there is no C# wrapper to native dll exist; also, having CLR version is a pretty smart solution and I like it.

ihedvall, 2025-02-07:
@sensboston 
I'm not a dotNET guru but I think it's related to the fact that the C++/CLR DLL is built against a specific platform (x64) and not as an "AnyCPU" platform like the other assemblies (DLL/EXE). I need to select a specific x64 platform when I'm running the unit tests. I think your application needs to be platform specific.

Costura have some special handling for embedding C++/CLR assemblies.
```xml
 <Costura>
  <UnmanagedWinX86Assemblies>
    Foo32
    Bar32
  </UnmanagedWinX86Assemblies>
  <UnmanagedWinX64Assemblies>
    Foo64
    Bar64
  </UnmanagedWinX64Assemblies>
  <UnmanagedWinArm64Assemblies>
    FooArm64
    BarArm64
  </UnmanagedWinArm64Assemblies>
</Costura>
```

It's possible to build a 32-bit x86 but I haven't prepared any Arm64 build in the mdflib.sln file.

I Google the error messages and it seems to be a common (unsolved) problem with C++/CLR loading.

sensboston, 2025-02-07:
@ihedvall 
I'm working in x64 mode **only** ('cause I need to manage a lot of data); I also googled this issue, with same results. 
OK, since my workaround working fine, I'll not waste more time for the issue, closing.

closed 2025-02-07 by sensboston