ihedvall/mdflib #117: Create an "MdfSortedWriter"

windfenggg, 2025-02-11:
Store samples in multiple DG blocks at the "same time."

ihedvall, 2025-02-20:
@windfenggg 
I have added support for multiple DG writes. There is no need to change your code but to speed up the write, you may change to the SaveSample() function that includes the DG block reference.

Note that there are some major changes internally why there might be some issues remaining. Note that the raw data (DT block) is divided into 4MB chunks (DL) and compressed (DT->DZ).

Please check if the changes works OK in your environment.

windfenggg, 2025-02-21:
@ihedvall 
   Thank you for your work, it's amazing! Today, I conducted a write test in my environment, and it can simultaneously write to multiple DG blocks without any issues so far.

ihedvall, 2025-02-28:
I close this issue. It is possible to reopen if any issues are found.

closed 2025-02-28 by ihedvall