ihedvall/mdflib #121: Source information missing for CreateCanConfig

tuna-f1sh, 2025-03-06:
First thanks for creating this lib. 

I'm still getting my head around mdf format and my C++ rusty (Rust pun not intended!) but I've been toying with a modified [`candump`](https://github.com/tuna-f1sh/can-utils/blob/mdflib/mdflib_c_wrapper.cpp) to log to MF4 format using this. Part of my toolchain for this is to process the MF4 with [python-can mf4 reader](https://github.com/hardbyte/python-can/blob/main/can/io/mf4.py).

Logging works fine and can be read with a viewer. The python-can module however checks for the source information to be source bus and type can bus: https://github.com/hardbyte/python-can/blob/main/can/io/mf4.py#L489

I could change that but think that this source information should be present? I noticed #120 and the [`BusConfigAdaptor`](https://github.com/ihedvall/mdflib/blob/main/mdflib/src/busconfigadapter.cpp) so perhaps you're already aware of this? I did make a fork and change which provides this: https://github.com/ihedvall/mdflib/compare/main...tuna-f1sh:mdflib:source

Are you planning to implement the `BusConfigAdaptor` to replace this `MdfWriter::CreateCanConfig` or does it make sense to include this as part of that function?

ihedvall, 2025-03-11:
@tuna-f1sh 
Sorry about the late response. GitHub didn't send me an e-mail notice?

Yes, I will move the CAN bus implementation so it is similar to the ETH and LIN messages. The CAN interface was done some years before the ETH and LIN implementations. The ETH and LIN introduced the Bus Config Adapter interface as well as introducing some basic Source Information block. 

The CAN configuration shall be moved to use the Bus Config Adapter interface as well but it currently using the older design. This include some extra properties as the bus and transport protocol in use. 

It's now possible to log several DG blocks in parallel. For a CAN bus this typical means a DG/CG block for each message ID. Unfortunately the ASAM standard naming of he CG blocks changes. It's never to easy.

I'm currently working with the new meta data (MD/HO) interface. It's actually pretty large change so it will take 1-2 weeks more to finalize. I will add the bus config task next.

tuna-f1sh, 2025-03-12:
No problem and thanks for the reply. 

I see it’s quite a re-work so no stress since I have that patch. Just glad to hear it’s on the task list. I’ll keep an eye out and help where I can.

ihedvall, 2025-05-20:
@tuna-f1sh 
The latest check-in now includes the source information. I have also added all 5 CAN, LIN, FlexRay, MOST and ETH bus interfaces. The multiple  DG support is added. It's now possible to set so-called filtered message. New IConfigAdapter interface added.

This is a major change regarding bus recording, so test before using it.

tuna-f1sh, 2025-05-20:
Great thanks for the tip. I'll give it a test soon.

closed 2025-08-02 by ihedvall