ihedvall/mdflib #122: MdfFileHistory/src/test.cpp  core dumped

flsh-bird, 2025-03-11:
when I compile and run the test.cpp, after write Sample datas, the following error occurr：
terminate called after throwing an instance of 'std::runtime_error'
  what():  MdfChannelGroup Init failed
Aborted (core dumped)

fail to execute "MdfChannelGroup can_data_frame = last_dg.GetChannelGroup("CAN_DataFrame");"

flsh-bird, 2025-03-11:
I compile and run the "test.cpp" on linux(arm64).

ihedvall, 2025-03-11:
@flsh-bird 
I think that the original error is found. It was an bus type enumerate that was mismatching in the DLL wrapper code. It was introduced when new bus types (LIN and ETH) was added. I have only tested in windows so you can check if it runs in arm64 now. 

Note that it's simpler to use the mdflib C++ library direct without the DLL wrapper code.

flsh-bird, 2025-03-12:
@ihedvall Thanks for your response, after modify the file(MdfExport.h), I get the same error:
terminate called after throwing an instance of 'std::runtime_error'
  what():  MdfChannelGroup Init failed
Aborted (core dumped)
The file path described in the title is not correct, I use "mdflibrary_example/src/test.cpp".

ihedvall, 2025-03-12:
@flsh-bird 
I think you did the same mistake as me, running the old DLL and not the new DLL with the modification. 

Skip the wrapper DLL and run the C++ library instead. The wrapper code doesn't handle that the channel group is missing (nullptr). It throws an exception instead which is bad error handling.

flsh-bird, 2025-03-12:
@ihedvall  thanks for your response again, I an a new mdf learner.
I compile the mdflibrary as a .so. I replace the .so before running the program in arm64.
In test.cpp, It use dg from 'Header.GetLastDataGroup()' to create cg direct, I dont know if its correct.
And I am trying to use mdflib direct.

flsh-bird, 2025-03-13:
@ihedvall Hi, Good afternoon~
I meet another error. I use tick_time to write samples into a .mf4 file. The tick_time is defined as follows.
And I use CANape to view the file, the time interval between data point is always 1s. It dose not work to modify "tick_time += 2000'000'000;".

`    auto tick_time = get_system_clock_ns();
    writer->StartMeasurement(tick_time);

    for (size_t sample = 0; sample < 100; ++sample) {
      auto value = static_cast<double>(sample) + 0.23;
      ch1->SetChannelValue(value);
      ch2->SetChannelValue(value);
      ch3->SetChannelValue(value);
      ch4->SetChannelValue(value);

      writer->SaveSample(*group1, tick_time);
      writer->SaveSample(*group2, tick_time);
      tick_time += 2000'000'000;
    }
    writer->StopMeasurement(tick_time);
    writer->FinalizeMeasurement();`

ihedvall, 2025-03-13:
@flsh-bird 
Have tested similar code with 2 seconds time between the samples and it work as expected.

I suspect that the above code is missing the (master) time channel (win64). Without the master channel, CANape doesn't know what to put on the X-axis. CANape should show sample number in this case.

Please let me know if it is something else than the missing time channel.

```c++
  auto* master2 = group2->CreateChannel();
  master2->Name("Time");
  master2->Type(ChannelType::Master);
  master2->Sync(ChannelSyncType::Time);
  master2->DataType(ChannelDataType::FloatBe);
  master2->DataBytes(8);
```

flsh-bird, 2025-03-13:
@ihedvall 
Sorry to bother you, I have fix my problem after adding master channel and setting the ChannelSyncType to time. >-<!

closed 2025-03-14 by ihedvall