ihedvall/mdflib #123: CreateCanConfig time channel naming

lawsonampere, 2025-03-11:
While applying DBC files using the [dbclib repo](https://github.com/ihedvall/dbclib), the DBC viewer produces an error about the time channel.

I'm not sure if the DBC viewer or the the mdflib is the actual issue, but since you have created both libraries I figured you may know better where a change is needed. Otherwise putting this here as it is an easy modification to make to the mdflib for those who need it. I simply changed the time channel created by the CreateCanConfig() function to be "Timestamp" instead of "t".

Modification of mdfwriter.cpp lines 603, 623, 632, and 650. Confirmed after this change that the DBC viewer can decode the file now.

[image]

Errors from DBC File Viewer with "t" time channel.
[image]

[image]

ihedvall, 2025-03-12:
@lawsonampere 
This is probably a bug in the DBC Viewer that assumes that the time channel name must be Timestamp. CSS log files uses Timestamp while Vector log  files uses t. The correct way of finding a time channel within the group is to searching for the channel marked as Master and with synchronization type equal to Time. 

I correct the bug in the DBC Viewer application. Note that there is an on-going change for CAN configuration that involves the SI blocks.

ihedvall, 2025-03-13:
I have changed the search for the time channels so it is independent of the channel name.

closed 2025-03-13 by ihedvall