ihedvall/mdflib #125: Memory consumption reading a MDF4 file with Signal Data blocks

daru91, 2025-03-13:
Hi, I have some mdf4 files containing uncompressed camera images in Signal Data (##SD) blocks. The issue is the memory consumption, it looks like the whole file contents is read once I create a Sample/Channel observer and read some samples using the partial data read function (see snippet below). Am I doing something wrong here or is this an issue with those SD blocks?

```
MdfReader reader("test.mf4");
reader.ReadEverythingButData();

auto header = reader.GetHeader();

auto dg = header->DataGroups()[0];
ISampleObserver sample_observer(*dg);

// Read 5 images (~60 MB). After this call memory usage increases to ~6GB (size of the file)
reader.ReadPartialData(*dg, 0, 5);
```

daru91, 2025-03-13:
@ihedvall 
I tried it with your suggested way, works like a charm at a decent speed without increasing the memory usage. 
Totally missed that Vlsd part in the documentation.

Thank you for your support and your efforts providing this library.

daru91, 2025-03-26:
I hope it's okay that I open this issue again, but is there a reason that the read function only works for MIME samples?

https://github.com/ihedvall/mdflib/blob/da8b4406d4ca2498907cb05af6d4795a3daa06b7/mdflib/src/mdfreader.cpp#L635C34-L635C61

Wouldn't it be possible to use it for `ChannelDataType::ByteArray` for example too?

ihedvall, 2025-03-26:
@daru91 
No, you are right. There is no need to test for the data type here. I guess the whole DataType() check can be removed. Variable length data is normally some sort of byte array. Note that the mime type for a MIME stream/sample is stored in its unit string. 

I will fix the bug.

ihedvall, 2025-04-02:
@daru91 
I have removed the MIME test. It shall now work for any byte array.

daru91, 2025-04-02:
@ihedvall Thank you for the update, while it works now, it is pretty slow (at least the way I use it). I have now a file with ~600000 messages (size of each message ~1KB). The way I'm using it currently is I'm just reading one sample.

I think the ReadCache/Dg4Block implementation is not designed for this approach (reading thousands of small samples one by one sequentially). It seems to me that the whole Block is processed to read one sample.

For the moment, I have adapted the code to reuse the same ReadCache when possible. 

Those are my adaptions:

dg4block.h

```
namespace mdf::detail {
class ReadCache; // forward declaration to avoid circular dependency

class Dg4Block : public DataListBlock, public IDataGroup {
...
private:
  ReadCache *read_cache_ = nullptr; 
...
}

```

dg4block.cpp

```
void  Dg4Block::ReadVlsdData(std::streambuf& buffer,Cn4Block& channel,
                  const std::vector<uint64_t>& offset_list,
                  std::function<void(uint64_t,
                              const std::vector<uint8_t>&)>& callback) {
...
  auto create_read_cache =
      [this](MdfBlock* block, std::streambuf& buffer, const std::vector<uint64_t>& offset_list,
         std::function<void(uint64_t, const std::vector<uint8_t>&)>& callback) {

        if (read_cache_) {
          if ((read_cache_->GetBlock() != block) || offset_list.empty() ||
                  (offset_list.front() != read_cache_->GetOffset())) {
            delete read_cache_;
            read_cache_ = nullptr;
          }
        }
        if (!read_cache_) { 
          read_cache_ = new ReadCache(block, buffer);
        }
        read_cache_->SetCallback(callback);
        read_cache_->SetOffsetFilter(offset_list);
      };

  if (read_signal_data) {
    create_read_cache(block, buffer, offset_list, callback);

    while (read_cache_->ParseSignalData()) {
    }
  } else if (read_vlsd_cg_data) {
    create_read_cache(this, buffer, offset_list, callback);

    read_cache_->SetRecordId(channel.VlsdRecordId());

    while (read_cache_->ParseVlsdCgData()) {
    }
  }
}
```

readcache.h

```
class ReadCache {
....

 uint64_t GetOffset() const;
 MdfBlock* GetBlock() const;

 private:
  MdfBlock* block_ = nullptr;

...
};
```

readcache.cpp

```
ReadCache::ReadCache( MdfBlock* block, std::streambuf& buffer)
: buffer_(buffer), block_(block) {
...
}

bool ReadCache::ParseSignalData() {
  if (data_count_ >= max_data_count_ || block_list_.empty()) {
    return false;
  }

  while (data_count_ < max_data_count_ ) {
    bool skip = false;
    try {

      std::vector<uint8_t> temp(4, 0);
      GetArray(temp);
      const LittleBuffer<uint32_t> length(temp, 0);

      if (!offset_filter_.empty() &&
           offset_filter_.find(offset_) == offset_filter_.cend()) {
        SkipBytes(length.value());
        skip = true;
        offset_ += 4 + length.value();
      } else {
        std::vector<uint8_t> raw_data(length.value());
        GetArray(raw_data);
        if (callback_) {
          callback_(offset_, raw_data);
        }
        offset_filter_.erase(offset_); // erase found offset
        offset_ += 4 + length.value();

        if (offset_filter_.empty())
        {
          return false; // return when all desired samples are found
        }
      }
    } catch (const std::exception &) {
      return false;
    }
    if (!skip) {
      break;
    }
  }
  return data_count_ <= max_data_count_;

}

```

This seems to work correctly with a decent speed for the files I need to process.
Probably a better approach for my issue would be to keep the current implementation and seek the correct position when the cache is created, but I couldn't get this running reliably.

So from my side for the moment everything is working fine, I'm not sure if my issue is in scope of this library, probably most users want to read the file at once and do some processing in the callback function.

ihedvall, 2025-04-03:
@daru91 
The MDF file format isn't great for read speed. You can optimize the SD read if the SD block only consist of a single DT block. In that circumstance, you only need to open the file and jump to the SD first offset + sample offset. Then just read the length and number of bytes and then close the file. There is no need for a read cache.

If the SD block is compressed, the SD block consist of  HL/DL/DZ blocks. Now it's much more messy. You need something as the current read cache but you can improve the cache a little bit as it isn't blended with other records (CG VLSD).

Before the ReadCache was introduced. the library opened a temporary file and did an uncompressed version of the DG/SD as in MDF version 3. It's now simple to parse the data samples. The disadvantage is that you basically read the data twice. Read from/to disc is costly.

For your problem, this might be a good idea. You read in the offsets in the first read and this should be quick. The next step is to create a temporary "SD" uncompressed file. The idea is to fetch the data bytes form this file with your knowledge of the offset. You need a "Save VLSD Data to File" function and a "Read VLSD Data from File". If we are lucky, the old code still works. 

1. Read the configuration as always
2. Read the sample offset list and timestamps.
3. Save the channel signal data to a temporary file (uncompressed data).
4. Open the temporary file, read the data and close it. Repeat until ready.

There exist some variation of the above solution as it resembles the Data Lake design with Parquet files but that may be to over-design the issue.

daru91, 2025-04-03:
@ihedvall 
Thank you for the proposals, I will try them if I face some performance issues again.

closed 2025-04-03 by daru91