ihedvall/mdflib #127: memory not release after call FinalizeMeasurement

flsh-bird, 2025-03-14:
@ihedvall
Hi, I wrote a testing program to record a .mf4 file. After calling the mf4_test3(), the rss memory do not decrease.
I run the test program in linux(arm64).
And in description of function CompressData(), how can i modify the setting( 4MB size or a 10 minute) to reduce the memory usage for compression.

1. CompressData() description 
If compression is used the data is saved at the 4MB size or a 10 minute max. If not using compression, the samples are saved each 10 second.).

2. memory result after calling mf4_test3() :
I0314 20:47:41.983722 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:42.981292 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:43.982326 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:44.982672 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:45.982722 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:46.982110 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:47.981881 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:48.982654 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:49.985467 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:50.983383 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:51.983421 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:52.982564 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s
I0314 20:47:53.984174 76174 source_monitor.cpp:248] cpu 0 %, rss 333756 KB, pss 329668 KB, io 0 kB_wr/s

3. test code:
`void mf4_test3() {

    auto writer = MdfFactory::CreateMdfWriter(MdfWriterType::Mdf4Basic);
    writer->Init("/app_data/mdf/test_sd_compress.mf4");
    writer->CompressData(true);
    auto* header = writer->Header();
    auto* history = header->CreateFileHistory();
    history->Description("Test data types");
    history->ToolName("MdfWrite");
    history->ToolVendor("ACME Road Runner Company");
    history->ToolVersion("1.0");
    history->UserName("Ingemar Hedvall");
    auto* dg = header->CreateDataGroup();
    std::vector<IChannelGroup *> cgs(80, nullptr);
    std::vector<IChannel *> cns(80, nullptr);
    for (size_t cg_num = 0; cg_num < 80; ++cg_num) {
        auto* cg = dg->CreateChannelGroup();
        cgs[cg_num] = cg;
        // master
        auto* master_ch = cg->CreateChannel();
        master_ch->Name("master");
        master_ch->Type(ChannelType::Master);
        master_ch->Sync(ChannelSyncType::Time);
        master_ch->DataType(ChannelDataType::FloatLe);
        master_ch->DataBytes(4);
        // channel
        auto* data_cn = cg->CreateChannel();
        std::string signal_name = "signal" +  std::to_string(cg_num);
        data_cn->Name(signal_name);
        data_cn->Type(ChannelType::FixedLength);
        data_cn->Sync(ChannelSyncType::None);
        data_cn->DataType(ChannelDataType::FloatLe);
        data_cn->DataBytes(4);
        cns[cg_num] = data_cn;
    }
    // start record
    writer->PreTrigTime(0);
    writer->InitMeasurement();
    auto tick_time = get_system_clock_ns();
    writer->StartMeasurement(tick_time);
    // record data
    std::random_device rd;  
    std::mt19937 gen(rd()); 
    std::uniform_int_distribution<> distrib(0, 1000);
    int randomNumber  = 0;

    for(size_t count = 0; count < 49800; ++count) {
        for (size_t cg_num = 0; cg_num < 80; ++cg_num) {
            randomNumber = distrib(gen);
            cns[cg_num]->SetChannelValue(randomNumber);
            writer->SaveSample(*cgs[cg_num], tick_time);
        }
        tick_time += 100'000'000;
        LOG(INFO)<<"--------------"<<+count;
    }
    writer->StopMeasurement(tick_time);
    writer->FinalizeMeasurement();
}

int main(int argc, char* argv[]) {
    signal(SIGINT, sigint);
    LOG(INFO)<<"--------------start record";
    mf4_test3();
    LOG(INFO)<<"--------------stop record";
    while(isRunning) {
        std::this_thread::sleep_for(std::chrono::milliseconds(5000));
    }

    return 0;
}`

ihedvall, 2025-03-14:
@flsh-bird 
Test Result in windows. Note that the peak Working Set is in the similar size as the MacOS RSS value.

Test without compression gives similar result (427 MB). If I add a small delay in the add sample loop, the peak size is 86MB.

```

BEFORE TEST
Page Faults: 2036
Peak Working Set: 7 MB
Working Set Size: 7 MB

[2025-03-14 18:14:33.057] [Trace] Save [::0]
[2025-03-14 18:14:33.057] [Trace] Finalize [::0]

AFTER TEST
Page Faults: 122856
Peak Working Set: 443 MB
Working Set Size: 10 MB
```

ihedvall, 2025-03-14:
Just some more log output with a small sleep in the add sample loop. This gives the write thread time to work.
```
BEFORE TEST
Page Faults: 2036
Peak Working Set: 7 MB
Working Set Size: 7 MB

[2025-03-14 18:28:56.818] [Trace] Waited for log file. Ticks: 10 [testwrite:void __cdecl mdf::test::TestWrite::SetUpTestSuite(void):167]
BEFORE INIT MEASUREMENT
Page Faults: 2255
Peak Working Set: 8 MB
Working Set Size: 8 MB

[2025-03-14 18:29:06.949] [Trace] Save [::0]
[2025-03-14 18:29:16.960] [Trace] Save [::0]
[2025-03-14 18:29:27.327] [Trace] Save [::0]
[2025-03-14 18:29:37.335] [Trace] Save [::0]
[2025-03-14 18:29:47.347] [Trace] Save [::0]
[2025-03-14 18:29:57.684] [Trace] Save [::0]
[2025-03-14 18:30:07.689] [Trace] Save [::0]
[2025-03-14 18:30:17.694] [Trace] Save [::0]
[2025-03-14 18:30:28.024] [Trace] Save [::0]
[2025-03-14 18:30:38.035] [Trace] Save [::0]
[2025-03-14 18:30:48.048] [Trace] Save [::0]
[2025-03-14 18:30:58.379] [Trace] Save [::0]
[2025-03-14 18:31:08.383] [Trace] Save [::0]
[2025-03-14 18:31:18.397] [Trace] Save [::0]
[2025-03-14 18:31:28.770] [Trace] Save [::0]
[2025-03-14 18:31:38.771] [Trace] Save [::0]
[2025-03-14 18:31:48.775] [Trace] Save [::0]
[2025-03-14 18:31:59.091] [Trace] Save [::0]
[2025-03-14 18:32:09.092] [Trace] Save [::0]
[2025-03-14 18:32:19.103] [Trace] Save [::0]
[2025-03-14 18:32:29.417] [Trace] Save [::0]
[2025-03-14 18:32:39.425] [Trace] Save [::0]
[2025-03-14 18:32:49.437] [Trace] Save [::0]
[2025-03-14 18:32:59.748] [Trace] Save [::0]
[2025-03-14 18:33:09.752] [Trace] Save [::0]
[2025-03-14 18:33:19.764] [Trace] Save [::0]
[2025-03-14 18:33:30.076] [Trace] Save [::0]
[2025-03-14 18:33:40.088] [Trace] Save [::0]
[2025-03-14 18:33:50.103] [Trace] Save [::0]
[2025-03-14 18:34:00.459] [Trace] Save [::0]
[2025-03-14 18:34:10.460] [Trace] Save [::0]
[2025-03-14 18:34:20.475] [Trace] Save [::0]
[2025-03-14 18:34:30.783] [Trace] Save [::0]
[2025-03-14 18:34:40.790] [Trace] Save [::0]
[2025-03-14 18:34:50.802] [Trace] Save [::0]
[2025-03-14 18:34:51.952] [Trace] Save [::0]
BEFORE STOP MEASUREMENT
Page Faults: 40769
Peak Working Set: 81 MB
Working Set Size: 80 MB

[2025-03-14 18:34:52.169] [Trace] Finalize [::0]

AFTER TEST
Page Faults: 41923
Peak Working Set: 82 MB
Working Set Size: 9 MB
```

flsh-bird, 2025-03-17:
@ihedvall 
Hi, thanks for your response. I have some questions about your response.
1. How to write the temporary cache manually during the process of writing into the same file.
"The test is little bit evil as you put 80 MB of sample data into the temporary cache before the write starts (test size each second). All is happening at the StopMeasurement(() call."
2. In my test, The rss memory is the real time memory consumption, not peak consumption. In your test, the memory reduce to "Working Set Size: 9 MB" after test. But my rss memory keeps 333MB after calling mf4_test3(), the program stays in the while loop .
AFTER TEST
Page Faults: 41923
Peak Working Set: 82 MB
Working Set Size: 9 MB

ihedvall, 2025-03-17:
@flsh-bird 
The internal queue solves 2 designs. 

The first is to be able to handle pre-trig time. When you initialize the measurement, The internal queue start accepting samples (SaveSample()). Nothing is stored to the file until you start the measurement. When starting the measurement, the samples between pre-trig time time and the start time can now be stored onto the file.

The second is to temporary store up to 4MB of samples so they can be compressed by the write thread. The write thread, checks for any samples to write, each 10 seconds.

The Mdf4Writer assumes that the sample arrive in real-time and that the measurement is longer than 10 seconds, giving the write thread time for at least one save. That's why I added a small delay in the sample test loop.

The above test is a very short test, all samples arrive with no delay. This means that all samples are stored in the internal queue and these samples are saved to the file when stopping the measurement.

The design with an internal queue and a write thread is questionable for a simple (embedded) logger. A simple logger opens the output file once and then streams all SaveSample() calls directly to the file. If you wants compression, you need the internal queue and you is back again with the memory problem.

There exist an MdfConverter writer type that doing some of the above. Unfortunately is mainly used to convert some existing file why compression is needed. I suppose a simpler version MdfSimpleLogger without compression is possible to do. The MdfReader fits all but the MdfWriters is dependent on there usage.

I can't help you with the memory issue, as I don't have a Linux Arm64/Arm32 development machine. I'm not sure which type of hardware that is recommended here (Raspberry?).

flsh-bird, 2025-03-18:
@ihedvall 
Hi, thanks for your response again.
I have solved my memory problem by using jemalloc to manage the memory. After calling mf4_test3, the rss memory reduce to 28.33M. But I do not find out the real reason - -.
1. In my usage scenario, I will try to make following adjustments:
a. WriteCache::WorkThread()
    Wake up the thread according to to sample number.
b. Writer4SampleQueue::SaveQueue
    process only one dz block even (QueueSize() > buffer_max)
2. DataBlock::ClearData(), it uses clear function to clear data, it can be improved.
    data_.clear()
3. One more  question, for Dz4Blocks/Dl4Blocks/Hl4Blocks, if these modules will be stored after call Writer4SampleQueue::SaveQueue. I haven't figured out their relationship yet.

ihedvall, 2025-03-18:
@flsh-bird 
Just minor comments.
1. I think that your memory consumption is related to the size of the internal sample queue. If you use compression, you need the queue to temporary store until you have enough data to compress. The QueueSize() function or its companion NofDzBlock() (> 1), is much faster now than earlier, so you can add a test in the SaveSample() function that triggers the write thread. There is a condition variable to set and I recommend to call std::this_thread::yield() to give the write thread a chance but unlock the (queue) mutex first.
2. I doubt that would do anything.
3. At start, only a HL block exist. The SaveQueue() will create a new DL + DZ for each call (NofDzBlock() > 1). When stopping the measurement, the last DL+DZ blocks are stored. This is a little bit messy but this is ASAM.

There is a MdfConverter writer type that keeps the MDF file open the whole time. I recommend to try your changes in that implementation. Converter application have similar memory issues as you have.

If you feel inspired, you may test with different queue list types (std::list, std::vector, std::deque).

flsh-bird, 2025-03-19:
@ihedvall 
Hi, for the second question from yesterday, when keeping writing in a same file :
1. All DL and DZ blocks created by CleanQueueCompressed() function are stored. And in DataBlock, it use "std::vector<uint8_t> data_" to store data. 'Data_.clear()' can not release the memory for vector container.
2. In CleanQueueCompressed(), when 'QueueSize() > buffer_max' is met, after the while loop, it may create a small dz/dt block to store data during the continuous writing process. If the QueueSize() is small, return the processing feels better.

ihedvall, 2025-03-19:
@flsh-bird 

1. Annoying. The std::vector::shrink_to_fit() should work but it's "not mandatory"? It is possible change to a unique_ptr instead but there will be some code changes. 
2. The function has a "bool finalize input argument" that defines if all samples should be stored. The initial approach was that SaveQueue() was called before stop of measurement and CleanQueue() was called after. I added a flag instead so I didn't have duplicate the code. Sorry about the design.

flsh-bird, 2025-03-19:
@ihedvall 
1.  swap function could be a good choice:
std::vector<uint8_t> new_data;
data_.swap(new_data);
2. only create a new dz/dt block when the buffer number is greater than 4'000'000, except the stop of measurement;
3. I have some new questions:
    a. Why need to store all dz and dl blocks, to write into mdf file after the stop of measurement?
    b. I still want to modify mf4writer to meet my needs, in function 'Writer4SampleQueue::CleanQueueCompressed()', if I keep the file open the whole time, what adjustments need to be made together to avoid casing problems.

flsh-bird, 2025-03-20:
@ihedvall 
Sorry, there is a 'convertersamplequeue', I do not notice it, no need to answer the question about MdfConverter.

ihedvall, 2025-03-20:
@flsh-bird 
Just for your knowledge. It exist an project item (issue) for the MDF writer for embedded system.
[Create MDF writer software for embedding](https://github.com/users/ihedvall/projects/1?pane=issue&itemId=48338647)
The current library is flexible but bulky. It assumes that you are running on a computer with a Linux/WIndows OS. 

I would recommend to leave the write thread as it is. The current solution works and doesn't add to the memory consumption. Fix some minor data.clear() issue. Just fix some vector::clear stuff. 

Focus on keeping the internal queue as small as possible. Add some extra check in the SaveSample() call, that triggers the write thread i.e. flushing the internal queue. Max queue size may be set by  a property on the writer (default 4MB). I would also change the std::deque based internal queue with a linked list (std::list) queue to gain better control over memory consumption. Maybe skipping the C++ containers!

In some test environment, the measurement may take a very long time (month). During this time, an external application, wants to periodically validates the samples as it is no meaning to continue. Therefore, do the write thread open and close the file. It also leaving the MDF file finalized.

For embedded systems, no external application can read the embedded file why  it's OK to keep it open and for the entire measurement. You can skip to update number of samples i.e. mark it as not finalized. Finalize the file when stopping the measurement.

ihedvall, 2025-03-25:
@flsh-bird 
Just for your information. 

The new MDF library for embedding is now planned for Zephyr RTOS.  [Embedding MDF](https://github.com/ihedvall/mdflib/discussions/129)

This library are some months in the future but it most likely will be a better base for embedded systems. Note write support only.