ihedvall/mdflib #144: Could you please tell me if there are any cases where C# was used to write for HL?

xinRedi, 2025-08-12:
[image]

ihedvall, 2025-08-26:
I made some unit test that created a compressed and uncompressed CAN bus log file (MDF).  I appended the unit test below. The C++/CLI wrapper mdflibrary.dll is compiled with the Visual Studio solution/project files not by CMAKE. There is a mdflib.sln file in the root of the repository. There exist some pre-built assemblies (32/64 bit) in the repository release area (mdflibrary.zip).

The HL-DL-DT/DZ data byte storage is used in 2 situations. When storing one measurement (DG) without compression, the data bytes can be appended directly onto the disc without delay. This technique is used by most embedded system loggers. If you want to minimize the disc size, compressing the data is an obvious option. Unfortunately you need to temporary store the data bytes (samples) temporarily in primary memory. The MDF standard state that the HL-DL-DZ storage should be used. The DZ block should not exceed 4 MB of data bytes.

The other situation is when doing multiple measurements in parallel. It's not possible to append data to the data file directly. The good thing with the HL-DL-DZ storage is that it is now possible to store multiple DG blocks in parallel. It is a bit tricky to implement and it requires temporary storage of the samples.

I'm not sure if I answered your question so please return with some more questions if needed. Note the the .NET assembly doesn't have all the C++ library functionality. 

```C#

[TestMethod]
public void TestCompressCanBus()
{
    MdfWriter writer = new MdfWriter(MdfWriterType.MdfBusLogger);
    string testFile = Path.Combine(_testDirectory, "compress_can.mf4");
    writer.Init(testFile);
    writer.CompressData = true;
    writer.MandatoryMemberOnly = true;    
    
    writer.BusType = MdfBusType.CAN;        
    writer.PreTrigTime = 0.0; 
    writer.CreateBusLogConfiguration();
        
    MdfHeader header = writer.Header;
    Assert.IsNotNull(header);
    
    // The file history is required in MDF 4
    MdfFileHistory fileHistory = header.CreateFileHistory();
    Assert.IsNotNull(fileHistory);
    fileHistory.Description = "Testing CAN bus logging with compression";
    fileHistory.ToolName = "MdfLibrary";
    fileHistory.ToolVendor = "IH Development";
    fileHistory.ToolVersion = "2.3.0";
    fileHistory.UserName = "Ingemar Hedvall";
    
    MdfDataGroup lastDataGroup = header.LastDataGroup;
    Assert.IsNotNull(lastDataGroup);

    MdfChannelGroup dataFrameGroup = lastDataGroup.GetChannelGroup("CAN_DataFrame");
    Assert.IsNotNull(dataFrameGroup);
    
    writer.InitMeasurement(); // Starts the internal cache
    ulong startTime = MdfLibrary.NowNs();
    writer.StartMeasurement(startTime);
    Assert.AreEqual(startTime, writer.StartTime);
    CanMessage msg = new CanMessage();
    msg.TypeOfMessage = CanMessageType.Can_DataFrame;
    
    msg.MessageId = 1234;
    msg.DataBytes = [ 1, 2, 3, 4, 5, 6, 7, 8 ];
    
    for (int sample = 0; sample < 1000; ++sample)
    {
        writer.SaveCanMessage(dataFrameGroup, startTime, msg); 
        startTime += 1_000_000_000u; // 1 s     
    }
    writer.StopMeasurement(startTime);
    writer.FinalizeMeasurement();
}

```

closed 2025-09-04 by ihedvall

xinRedi, 2025-09-15:
> 我做了一些单元测试，创建了压缩和未压缩的 CAN 总线日志文件 （MDF）。我在下面附加了单元测试。C++/CLI 包装器mdflibrary.dll是使用 Visual Studio 解决方案/项目文件编译的，而不是由 CMAKE 编译的。存储库的根目录中有一个 mdflib.sln 文件。存储库发布区域 （mdflibrary.zip） 中存在一些预构建的程序集（32/64 位）。
> 
> HL-DL-DT/DZ 数据字节存储用于 2 种情况。当存储一个测量值 （DG） 而不进行压缩时，数据字节可以毫不延迟地直接附加到光盘上。大多数嵌入式系统记录仪都使用这种技术。如果您想最小化光盘大小，压缩数据是一个显而易见的选择。不幸的是，您需要将数据字节（样本）临时存储在主内存中。MDF 标准规定应使用 HL-DL-DZ 存储。DZ 块的数据字节不应超过 4 MB。
> 
> 另一种情况是并行进行多个测量时。无法将数据直接追加到数据文件。HL-DL-DZ 存储的好处是现在可以并行存储多个 DG 块。实现起来有点棘手，需要临时存储样品。
> 
> 我不确定我是否回答了您的问题，因此如果需要，请带着更多问题返回。请注意，.NET 程序集不具有所有 C++ 库功能。
> 
> [TestMethod]
> public void TestCompressCanBus()
> {
>     MdfWriter writer = new MdfWriter(MdfWriterType.MdfBusLogger);
>     string testFile = Path.Combine(_testDirectory, "compress_can.mf4");
>     writer.Init(testFile);
>     writer.CompressData = true;
>     writer.MandatoryMemberOnly = true;    
>     
>     writer.BusType = MdfBusType.CAN;        
>     writer.PreTrigTime = 0.0; 
>     writer.CreateBusLogConfiguration();
>         
>     MdfHeader header = writer.Header;
>     Assert.IsNotNull(header);
>     
>     // The file history is required in MDF 4
>     MdfFileHistory fileHistory = header.CreateFileHistory();
>     Assert.IsNotNull(fileHistory);
>     fileHistory.Description = "Testing CAN bus logging with compression";
>     fileHistory.ToolName = "MdfLibrary";
>     fileHistory.ToolVendor = "IH Development";
>     fileHistory.ToolVersion = "2.3.0";
>     fileHistory.UserName = "Ingemar Hedvall";
>     
>     MdfDataGroup lastDataGroup = header.LastDataGroup;
>     Assert.IsNotNull(lastDataGroup);
> 
>     MdfChannelGroup dataFrameGroup = lastDataGroup.GetChannelGroup("CAN_DataFrame");
>     Assert.IsNotNull(dataFrameGroup);
>     
>     writer.InitMeasurement(); // Starts the internal cache
>     ulong startTime = MdfLibrary.NowNs();
>     writer.StartMeasurement(startTime);
>     Assert.AreEqual(startTime, writer.StartTime);
>     CanMessage msg = new CanMessage();
>     msg.TypeOfMessage = CanMessageType.Can_DataFrame;
>     
>     msg.MessageId = 1234;
>     msg.DataBytes = [ 1, 2, 3, 4, 5, 6, 7, 8 ];
>     
>     for (int sample = 0; sample < 1000; ++sample)
>     {
>         writer.SaveCanMessage(dataFrameGroup, startTime, msg); 
>         startTime += 1_000_000_000u; // 1 s     
>     }
>     writer.StopMeasurement(startTime);
>     writer.FinalizeMeasurement();
> }

Thank you for providing the detailed context and unit tests. I appreciate the clarification regarding the HL-DL-DT/DZ data byte storage for both compressed and uncompressed CAN bus log files (MDF), as well as the explanation about the trade-offs between direct disk appending and memory buffering for compression and parallel measurements.  

Your points about the benefits of HL-DL-DZ storage for handling multiple parallel measurements and the 4 MB data byte limit for DZ blocks are clear. This addresses my questions about the implementation nuances and temporary sample storage requirements.  

Also, thank you for confirming that the C++/CLI wrapper (mdflibrary.dll) is compiled using Visual Studio (mdflib.sln) rather than CMAKE, and for noting the availability of pre-built assemblies in the repository's release area.  

If I have further questions as I continue working with the library, I’ll be sure to reach out. Thanks again for your help!  

Best regards