ihedvall/mdflib #152: ReadCache::ParseSignalData() enhancement

TimLinProg, 2025-09-17:
### Backgroud
We have record data in mdf4 format, the size of one mdf4 file would be dozen of GB, e.g. 40GB, millions of VLSD samples in it, and we use MdfReader::ReadVlsdData() with offset_list and callback function to read desired samples (all or partial depends on user input).

### Founding
We found ReadVlsdData() take even longer when reading partial of samples (e.g. only 10K samples) than when reading all the samples (millions).

### Investigation
Below are what I understanding on ReadCache::ParseSignalData(), please correct me if I was wrong.
- offset_filter_ is a set in assending order filled with input offset_list
- offset matching walk from first signal data to last signal data, no matter if the offset_ is greater than the maximum offset in offset_list_

```
bool ReadCache::ParseSignalData() {
  if (data_count_ >= max_data_count_ || block_list_.empty()) {
    return false;
  }

  while (data_count_ < max_data_count_ ) {
    bool skip = false;
    try {

      std::vector<uint8_t> temp(4, 0);
      GetArray(temp);
      const LittleBuffer<uint32_t> length(temp, 0);

      if (!offset_filter_.empty() &&
           offset_filter_.find(offset_) == offset_filter_.cend()) {
        SkipBytes(length.value());
        skip = true;
        offset_ += 4 + length.value();
      } else {
        std::vector<uint8_t> raw_data(length.value());
        GetArray(raw_data);
        if (callback_) {
          callback_(offset_, raw_data);
        }
        offset_ += 4 + length.value();
      }
    } catch (const std::exception &) {
      return false;
    }
    if (!skip) {
      break;
    }
  }
  return data_count_ <= max_data_count_;
}
```

### Ehancement
Below might be ehancement point:
- After experiment, we found SkipBytes(length.value()) takes more time than GetArray(raw_data) with our data, I do not know why yet.
- return false  if the offset_ is greater than the maximum offset in offset_list_, e.g.
```
         if (!offset_filter_.empty() &&
           offset_filter_.find(offset_) == offset_filter_.cend()) {
        SkipBytes(length.value());
        skip = true;
        offset_ += 4 + length.value();
        
        if (offset_ > *offset_filter_.rbegin()) {
            /*reach end of offset_list*/
            return false;
        }
      }
```

- Suggest to change ReadVlsdData callback function to "std::function<**bool**(uint64_t, const std::vector<uint8_t>&)>", when callback return false ReadVlsdData abort the processing, which give opportunity to upper user (application) to terminate in case of something goes wrong, e.g sample data corrupiton,etc, for instance.
```
        std::vector<uint8_t> raw_data(length.value());
        GetArray(raw_data);
        if (callback_) {
            //callback_(offset_, raw_data)
            if (!callback_(offset_, raw_data))
            {
                //user reqeust to abort
                return false;
            }
        }
        offset_ += 4 + length.value(); 
```

- Given offset_filter_ is a set in assending order, could offset_ directly move to start of Data Link (e.g. the first signal data correpsonding to offset_list is under Data Link 5) to speed up first match?

Thanks,
Tim

ihedvall, 2025-09-18:
@TimLinProg 
The Read VLSD Data function was designed to solve the problem of large files where each sample includes a video stream. The test file was a 23GB file with 1000 samples. The callback solves the memory allocation problem as it only needs to allocate one sample (offset). Note that there is a one-to-one relation between sample and offset but there might be less unique offsets than samples.

Reading the VLSD data is split into 2 parsers, CG-VSLD and SD-VLSD. The CG-VLSD parser needs to read from the beginning of the DG's data block and call the callbacks for each offset. It can abort when the last offset is found.

Reading SD-VLSD data is simpler until some evil logger splits offset data into two SD block. Read SD VLSD data can be improved according to your proposals.

1. Skipping bytes should be faster than reading bytes but I had some strange happenings when reading from a NAS drive. You can test this by comparing the test reading from a local drive (block mode). I did a test of reading from a NAS (SMB) drive or using the iSCSI protocol and the iSCSI was significant faster (~2x). Note same NAS and network.
2. Returning bool is added to some callbacks. Changing a signature here shouldn't be a big issue but other user may complains.
3. The (internal ReadCache) offset filter is a std::set<uint64_t>. It's possible to optimize the read by skipping to the right DL/SD block(s) and repeat for the next offset.

The [1], [2] and [3] points are fairly easy to implement but verify the functions are much harder without some test files.

TimLinProg, 2025-09-20:
> [@TimLinProg](https://github.com/TimLinProg) I have found a Vector SD List file that I can verify the changes on. It's not so big so performance testing can be tricky to evaluate.

Great, looking forward to your update. BTW, our file is recorded by Vector's device.

closed 2025-09-26 by ihedvall